mvp/uhubctl #481: Can't cycle port with supported HUB with a raspberry pi 2B

HoracioDos, 2023-01-02:
Hello! First I want to thank you for this software!

I own a TpLink usb 3.0 Hub (4 ports) model UH400 Hwd Version 4.0. Uhubctl works fine when this hub is connected to a notebook with Linux Mint 21.1 Kernel 5.15.0.56 and uhubctl version 2.5.0-2-g19dfdad7
```
sudo uhubctl 
Current status for hub 2-2 [0bda:0411 Generic USB3.2 Hub, USB 3.20, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 1-5 [0bda:5411 Generic USB2.1 Hub, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
```
```
wHubCharacteristic 0x00a9
    Per-port power switching
    Per-port overcurrent protection
    TT think time 16 FS bits
```
So you can add this model to the compatibility table

But.... When I plug it to a raspberry pi 2B, uhubctl can't cycle any port in this hub. I don't know if it's due that this hub is usb 3.0 and raspberry pi 2B is only usb 2.0. I created both udev rules without success. I also plugged it in every port (1-4). Any idea about this?

I also own a Targus ACH214 (4 ports) USB HUB 2.0 and uhubctl can detect and cycle ports with force parameter. It is semi-compatible.
```
Current status for hub 1-5 [05e3:0608 USB2.0 Hub, USB 2.00, 4 ports, ganged]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
```
```
  wHubCharacteristic 0x00e0
    Ganged power switching
    Ganged overcurrent protection
    TT think time 32 FS bits
    Port indicators
```
but again I can't cycle ports when it is connected to the RP2B. I have a third USB Hub 3.0 with a Via Chip and it seems to be same case as the Targus but I didn't fully tested.

Thanks!!

mvp, 2023-01-02:
USB3 devices have 2 circuits inside of them: USB3 and USB2. To turn power off, uhubctl sends commands to both of them (if hub is connected via USB3). When hub connected as USB2 only (e.g. host is USB2 only of cable is USB2 only), then while most hub chipsets support it properly and turn power off even if on USB2 only, some have bad firmware which fails to control USB3 part, and that makes it not work when connected via USB2.

That said, when you say that it works for you on USB3, that is not 100% obvious. To ensure that please do vbus test - check that connected phone stops charging or USB light stops shining. Ideally, test voltage with multimeter.

HoracioDos, 2023-01-10:
I'll try to follow your instructons this weekend and I'll report back

mvp, 2023-01-27:
Closing by inactivity. Please reopen if you think this is still a problem.

closed 2023-01-27 by mvp