mvp/uhubctl #486: Asus Sabertooth z77 motherboard confirmed working

Asherathe, 2023-01-17:
Current status for hub 5-1 [8087:0024, USB 2.00, 8 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
  Port 5: 0100 power
  Port 6: 0100 power
  Port 7: 0100 power
  Port 8: 0100 power
Current status for hub 3-1 [8087:0024, USB 2.00, 6 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0503 power highspeed enable connect [05ac:12a8 Apple Inc. iPhone 3d2c02406ad6b879cf693eb37f5fd84bb0e44917]
  Port 4: 0100 power
  Port 5: 0100 power
  Port 6: 0100 power

I was able to turn power to the iphone off and on, so I believe this is compatible.

mvp, 2023-01-19:
Can you please check if vbus is turned off? All too often, this does not work on motherboard's onboard USB hubs. Just check if phone stops charging or USB fan stops spinning (or USB light stops shining).

Asherathe, 2023-01-19:
I already said I was able to turn power to the iphone off and on. What else is needed to consider it working?

mvp, 2023-01-27:
Thanks for reporting, added to the table.

closed 2023-01-27 by mvp