mvp/uhubctl #488: UGREEN USB Hub 3.0 50985 confirmed working

patrickwasp, 2023-01-26:
UGREEN GROUP LIMITED 50985
https://www.ugreen.com/en-ca/products/4-ports-usb-3-0-hub?variant=39915659165758

```
Current status for hub 2-1 [2109:0817 VIA Labs, Inc. USB3.0 Hub 000000000, USB 3.20, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 1-2 [2109:2817 VIA Labs, Inc. USB2.0 Hub 000000000, USB 2.10, 4 ports, ppps]
  Port 1: 0103 power enable connect [2516:0011 CM Storm Quickfire TKL 6keys]
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
```

mvp, 2023-01-27:
Can you please check for vbus off support?
https://github.com/mvp/uhubctl#how-do-i-check-if-my-usb-hub-is-supported-by-uhubctl
All too often, hubs do not actually turn voltage off.
Thanks!

patrickwasp, 2023-01-27:
I can turn each port USB3 port off while leaving the others on. This is the output when all are off:

```
Current status for hub 2-1 [2109:0817 VIA Labs, Inc. USB3.0 Hub 000000000, USB 3.20, 4 ports, ppps]
  Port 1: 00a0 off
  Port 2: 00a0 off
  Port 3: 00a0 off
  Port 4: 00a0 off
Current status for hub 1-2 [2109:2817 VIA Labs, Inc. USB2.0 Hub 000000000, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0103 power enable connect [2516:0011 CM Storm Quickfire TKL 6keys]
```

The backlights on the keyboard turn off for a few seconds when I turn it off but come back up later on their own even though it says that the USB power is off. This hub only has 4 USB 3 ports but I'm guessing it has 4 virtual USB 2 ports that don't seem to respond correctly. I guess this means that only the "cycle" option actually works.

There is also a mislabeling of the USB ports. 
Couting from the side of the hub with the cable connecting to the PC:
| Physical port | uhubct port |
|---------------|-------------|
| 1             | 3           |
| 2             | 2           |
| 3             | 1           |
| 4             | 4           |

mvp, 2023-01-28:
> The backlights on the keyboard turn off for a few seconds when I turn it off but come back up later on their own even though it says that the USB power is off. This hub only has 4 USB 3 ports but I'm guessing it has 4 virtual USB 2 ports that don't seem to respond correctly. I guess this means that only the "cycle" option actually works without hacks.

That does not seem like device we would want to add to the compatible table.
Compatible devices should fully work including turning voltage off when requested.

Note that USB2/USB3 ports do not exist on their own - they are inherently paired together to their single physical port - read more here https://github.com/mvp/uhubctl#usb-30-duality-note. So for USB3 hub both USB2 and USB3 components must work properly(unless device is not USB statandards compliant, or OS bug, or uhubctl bug). All USB3 devices currently listed in supported table are proven to work correctly with this regard.

However, since you mention that your port power is flashing momentarily, you may be hitting Linux issue https://github.com/mvp/uhubctl#power-comes-back-on-after-few-seconds-on-linux. There are few workarounds listed in that FAQ article, try them. Or, try it on Mac or FreeBSD - they do not have this issue.

patrickwasp, 2023-01-28:
Yeah the repeat hack works but I you're right this one doesn't seem a good fit.

closed 2023-01-28 by patrickwasp