mvp/uhubctl #489: SHA Mismatch while running `brew install`

liberato-whisper, 2023-01-30:
Running the recommended install for macOS isn't working locally for me:

```
% brew tap mvp/uhubctl https://github.com/mvp/uhubctl

==> Tapping mvp/uhubctl
Cloning into '/usr/local/Homebrew/Library/Taps/mvp/homebrew-uhubctl'...
remote: Enumerating objects: 917, done.
remote: Counting objects: 100% (293/293), done.
remote: Compressing objects: 100% (121/121), done.
remote: Total 917 (delta 176), reused 201 (delta 159), pack-reused 624
Receiving objects: 100% (917/917), 245.61 KiB | 4.39 MiB/s, done.
Resolving deltas: 100% (561/561), done.
Tapped 1 formula (21 files, 366.8KB).
% brew install uhubctl

==> Fetching mvp/uhubctl/uhubctl
==> Downloading https://github.com/mvp/uhubctl/archive/v2.5.0.tar.gz
Already downloaded: /Users/generic-user/Library/Caches/Homebrew/downloads/ab52b9ee5f8836cd159b02e71fa74cee5ae01ca941fb78e1c99bfc79c314d6cb--uhubctl-2.5.0.tar.gz
Error: uhubctl: SHA256 mismatch
Expected: d4452252f7862f7a45dd9c62f2ea7cd3a57ab5f5ab0e54a857d4c695699bbba3
  Actual: 7be75781b709c36c03c68555f06347d70e5f4e8fd2d17fd481f20626fb4c6038
    File: /Users/generic-user/Library/Caches/Homebrew/downloads/ab52b9ee5f8836cd159b02e71fa74cee5ae01ca941fb78e1c99bfc79c314d6cb--uhubctl-2.5.0.tar.gz
To retry an incomplete download, remove the file above.
```

mvp, 2023-01-30:
It works for me now:
```
$ brew tap mvp/uhubctl https://github.com/mvp/uhubctl
$ brew update
Updated 1 tap (mvp/uhubctl).

$ brew install uhubctl
==> Fetching mvp/uhubctl/uhubctl
==> Downloading https://github.com/mvp/uhubctl/archive/v2.5.0.tar.gz
==> Installing uhubctl from mvp/uhubctl
==> make
🍺  /usr/local/Cellar/uhubctl/2.5.0: 6 files, 99KB, built in 5 seconds
$ uhubctl -v
2.5.0
```

closed 2023-01-30 by mvp

liberato-whisper, 2023-01-30:
Thanks for the quick turnaround, working here too!

liberato-whisper, 2023-01-31:
```
% brew reinstall uhubctl
==> Fetching mvp/uhubctl/uhubctl
==> Downloading https://github.com/mvp/uhubctl/archive/v2.5.0.tar.gz
==> Downloading from https://codeload.github.com/mvp/uhubctl/tar.gz/refs/tags/v2.5.0
######################################################################## 100.0%
Error: SHA256 mismatch
Expected: 7be75781b709c36c03c68555f06347d70e5f4e8fd2d17fd481f20626fb4c6038
  Actual: d4452252f7862f7a45dd9c62f2ea7cd3a57ab5f5ab0e54a857d4c695699bbba3
    File: /Users/generic-user/Library/Caches/Homebrew/downloads/ab52b9ee5f8836cd159b02e71fa74cee5ae01ca941fb78e1c99bfc79c314d6cb--uhubctl-2.5.0.tar.gz
To retry an incomplete download, remove the file above.
```

seems to have returned and reverted to the old hash?

commit 61fd84b0 (2023-01-30): Fixed brew sha256 for v2.5.0.tar.gz

This used to work fine, perhaps something has changed in how Github
creates archives so archive checksum does not match anymore?
Confirmed that actual file content at https://github.com/mvp/uhubctl/archive/v2.5.0.tar.gz
is exactly the same as content for `git checkout v2.5.0`.
Updated checksum to expected value so `brew install uhubctl` should pass.
Closes #489.