mvp/uhubctl #491: What did I do with Raspberry Pi?

dims12, 2023-02-02:
I had my Raspi 3B+ running old raspbian Jessy. I was connecting to it by ssh. I have installed [uhubctl](https://github.com/mvp/uhubctl) and got

```
# uhubctl
Current status for hub 1-1 [0424:9514, USB 2.00, 5 ports, ppps]
  Port 1: 0503 power highspeed enable connect [0424:ec00]
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
  Port 5: 0100 power
Current status for hub 1 [1d6b:0002 Linux 4.14.50-v7+ dwc_otg_hcd DWC OTG Controller 3f980000.usb, USB 2.00, 1 ports, ppps]
  Port 1: 0503 power highspeed enable connect [0424:9514, USB 2.00, 5 ports, ppps]
```
Then I did

```
# uhubctl -a off -l 1-1 -p 1
Current status for hub 1-1 [0424:9514, USB 2.00, 5 ports, ppps]
  Port 1: 0503 power highspeed enable connect [0424:ec00]
```
and lost access to my Raspi.

What did I do? Powered off my network card or what?

mvp, 2023-02-02:
This cannot be Raspberry Pi 3B+ - it has different USB topology https://github.com/mvp/uhubctl#raspberry-pi-3b.

Your topology corresponds to Raspberry Pi 2B or 3B, which have 5 USB ports on one hub 1-1, first port being network (both wifi and Ethernet), and ports 2-5 all controlled by port 2 https://github.com/mvp/uhubctl#raspberry-pi-b2b3b.

mvp, 2023-02-02:
Basically, don't turn network off unless you want to save some power and don't need network connection. To recover from this, simply reset your Pi power.

closed 2023-02-02 by mvp

dims12, 2023-02-19:
Yes, sorry my revision is `a02082`.

How to turn off correct device then?

mvp, 2023-02-19:
Please readme for Raspberry Pi. Note that for 2B or 3B you can control port 2 to turn off power on ports 2-5 all ot once. Unfortunately, RPI hardware doesn't support controlling power per port - nothing can do that, all ports are powered from the same source. Port 1 controls network, and is independent from other ports.