mvp/uhubctl #493: CH334H usb 2.0 hub ( 1a86:8091 QinHeng Electronics ) support

nikitos1550, 2023-02-13:
Hello! Does this hub ( 1a86:8091 QinHeng Electronics ) supported?

nikitos1550, 2023-02-13:
I tried force mode
```
nikita1550@nikita1550-nix ~/g/t/t/uhubctl> sudo ./uhubctl -f -l 1-1 -p 2 -a 0
Current status for hub 1-1 [1a86:8091 USB HUB, USB 2.00, 4 ports, ganged]
  Port 2: 0000 off
Sent power off request
New status for hub 1-1 [1a86:8091 USB HUB, USB 2.00, 4 ports, ganged]
  Port 2: 0000 off
nikita1550@nikita1550-nix:/home/nikita1550/gohw/tmp3/tmp/uhubctl$ 
nikita1550@nikita1550-nix ~/g/t/t/uhubctl> sudo ./uhubctl -f -l 1-1 -p 2 -a 1
Current status for hub 1-1 [1a86:8091 USB HUB, USB 2.00, 4 ports, ganged]
  Port 2: 0000 off
Sent power on request
New status for hub 1-1 [1a86:8091 USB HUB, USB 2.00, 4 ports, ganged]
  Port 2: 0100 power
```
I use self made hub pcba. So seems in terms of software it is ok, just schematics need to be updated.

mvp, 2023-02-14:
Sorry, we do not consider hubs that require -f as compatible.
Also, it you made this hub yourself, that is not something average person can buy.

closed 2023-02-14 by mvp