mvp/uhubctl #496: VL817 ignores state change for USB3 ports

kirill9617, 2023-03-25:
I have noname VL817 hub that I modified to support vbus control.
But power switching works only if hub is connected to USB2 uptream only.
For USB3 connection power off command is ignored.

Log with no devices connected:
`sudo uhubctl -l 4-3 -p 2 -a off`
```
Current status for hub 4-3 [2109:0817 VIA Labs, Inc. USB3.0 Hub, USB 3.10, 4 ports, ppps]
  Port 2: 02a0 power 5gbps Rx.Detect
Sent power off request
New status for hub 4-3 [2109:0817 VIA Labs, Inc. USB3.0 Hub, USB 3.10, 4 ports, ppps]
  Port 2: 02a0 power 5gbps Rx.Detect
Current status for hub 1-3 [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 2: 0100 power
Sent power off request
New status for hub 1-3 [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 2: 0000 off
```

Log with USB3 device connected (device was reseted, but stays powered on):
`sudo uhubctl -l 4-3 -p 2 -a off`
```
Current status for hub 4-3 [2109:0817 VIA Labs, Inc. USB3.0 Hub, USB 3.10, 4 ports, ppps]
  Port 2: 0203 power 5gbps U0 enable connect [13fe:5200 USB DISK 3.0 07083BB424907020]
Sent power off request
New status for hub 4-3 [2109:0817 VIA Labs, Inc. USB3.0 Hub, USB 3.10, 4 ports, ppps]
  Port 2: 0203 power 5gbps U0 enable connect [13fe:5200]
Current status for hub 1-3 [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 2: 0100 power
Sent power off request
New status for hub 1-3 [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 2: 0000 off
```

Log with USB2 device connected (device was logically disconnected, but stays powered):
`sudo uhubctl -l 4-3 -p 2 -a off`
```
Current status for hub 4-3 [2109:0817 VIA Labs, Inc. USB3.0 Hub, USB 3.10, 4 ports, ppps]
  Port 2: 02a0 power 5gbps Rx.Detect
Sent power off request
New status for hub 4-3 [2109:0817 VIA Labs, Inc. USB3.0 Hub, USB 3.10, 4 ports, ppps]
  Port 2: 02a0 power 5gbps Rx.Detect
Current status for hub 1-3 [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 2: 0103 power enable connect [10c4:ea60 Silicon Labs CP2104 USB to UART Bridge Controller 018AD9EF]
Sent power off request
New status for hub 1-3 [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 2: 0000 off
```

mvp, 2023-03-30:
This is not a VL817 chip issue, but rather hub manufacturer problem - most likely they did not wire things properly, or saved few cents on some circuitry.

Since this is noname hub and doesn't really work, there is no point in adding it to supported table.

Please check hubs in supported table, they should all work correctly. Few of them use Via chips too.

closed 2023-03-30 by mvp

kirill9617, 2023-06-16:
The problem was in VL817 firmware. After adding SPI flash with VL817_Q7_9033 flashed to it hub works as expected with USB3 upstream. Maybe it will help somebody else.

matwey, 2026-10-18:
I think I have the same issue. Do you have VL817-Q7 B0 revision or C0 revision chip?