mvp/uhubctl #498: Feature Request:  Control power to USB PD / Thunderbolt / USB4 devices

sslupsky, 2023-03-29:
It would be helpful if this tool could be enhanced to control power delivery for Type C host ports that support USB PD so that we can control USB 3.1, Thunderbolt and USB4 devices.

mvp, 2023-03-30:
USB4 hubs should be supported already, as long as they advertise per port power switching. Per USB 3.2 spec, any hub that has USB-C downstream ports must support it, and afaik all USB4 hubs must have only USB-C downstream ports.

That said, I have not seen USB4 hubs yet. Afaik, the only true USB4 hub available on the market is Caldigit Element Hub, but I don't have it, and don't feel like splurging $250 to get this toy.

As far as I know, USB PD protocol is completely different compared to what uhubctl does. I would love to understand how it could be done, but I don't really have enough information on how to make it happen, yet.

mvp, 2026-06-06:
It does not seem like true PD control is even possible from user space.
I am closing this. May reopen if anything else comes up.
At any rate, PD works completely differently, and software PD control is rarely accessible from user space.
When software control is available (very rare!), vendors use obscure undocumented custom protocols.