mvp/uhubctl #499: brew fails

sslupsky, 2023-03-29:
I attempted to install uhubctl using brew and encountered a build error.  I have libusb installed (using brew).
macOS 11.7.5 (Big Sur)

brew info libusb:
```
barbie-en0:~ stevenslupsky$ brew info libusb
==> libusb: stable 1.0.26 (bottled), HEAD
Library for USB device access
https://libusb.info/
/usr/local/Cellar/libusb/1.0.26 (22 files, 532KB)
  Poured from bottle on 2022-08-26 at 16:56:12
From: https://github.com/Homebrew/homebrew-core/blob/HEAD/Formula/libusb.rb
License: LGPL-2.1-or-later
==> Options
--HEAD
	Install HEAD version
==> Analytics
install: 2,807 (30 days), 164,107 (90 days), 1,240,909 (365 days)
install-on-request: 357 (30 days), 17,808 (90 days), 123,506 (365 days)
build-error: 1 (30 days)
```

brew install uhubctl:
```
barbie-en0:~ stevenslupsky$ brew install uhubctl
==> Fetching mvp/uhubctl/uhubctl
==> Downloading https://github.com/mvp/uhubctl/archive/v2.5.0.tar.gz
Already downloaded: /Users/stevenslupsky/Library/Caches/Homebrew/downloads/ab52b9ee5f8836cd159b02e71fa74cee5ae01ca941fb78e1c99bfc79c314d6cb--uhubctl-2.5.0.tar.gz
==> Installing uhubctl from mvp/uhubctl
==> make
Last 15 lines from /Users/stevenslupsky/Library/Logs/Homebrew/uhubctl/01.make:
2023-03-29 21:14:44 +0000

make

fatal: not a git repository (or any of the parent directories): .git
clang  -g -O0 -Wall -Wextra -std=c99 -pedantic -DPROGRAM_VERSION=\"2.5.0\" -I/usr/local/Cellar/libusb/1.0.26/include/libusb-1.0 uhubctl.c -o uhubctl -L/usr/local/Cellar/libusb/1.0.26/lib -lusb-1.0
uhubctl.c:35:10: fatal error: 'libusb-1.0/libusb.h' file not found
#include <libusb-1.0/libusb.h>
         ^~~~~~~~~~~~~~~~~~~~~
1 error generated.
make: *** [uhubctl] Error 1

If reporting this issue please do so to (not Homebrew/brew or Homebrew/homebrew-core):
  mvp/uhubctl

/usr/local/Homebrew/Library/Homebrew/utils/github/api.rb:315:in `raise_error': Validation Failed: [{"message"=>"The listed users and repositories cannot be searched either because the resources do not exist or you do not have permission to view them.", "resource"=>"Search", "field"=>"q", "code"=>"invalid"}] (GitHub::API::ValidationFailedError)
	from /usr/local/Homebrew/Library/Homebrew/utils/github/api.rb:241:in `open_rest'
	from /usr/local/Homebrew/Library/Homebrew/utils/github.rb:184:in `search'
	from /usr/local/Homebrew/Library/Homebrew/utils/github.rb:188:in `search_results_items'
	from /usr/local/Homebrew/Library/Homebrew/utils/github.rb:38:in `search_issues'
	from /usr/local/Homebrew/Library/Homebrew/utils/github.rb:68:in `issues_for_formula'
	from /usr/local/Homebrew/Library/Homebrew/exceptions.rb:498:in `fetch_issues'
	from /usr/local/Homebrew/Library/Homebrew/exceptions.rb:493:in `issues'
	from /usr/local/Homebrew/Library/Homebrew/exceptions.rb:549:in `dump'
	from /usr/local/Homebrew/Library/Homebrew/brew.rb:151:in `rescue in <main>'
	from /usr/local/Homebrew/Library/Homebrew/brew.rb:139:in `<main>'
/usr/local/Homebrew/Library/Homebrew/formula.rb:2543:in `block in system': Failed executing: make (BuildError)
	from /usr/local/Homebrew/Library/Homebrew/formula.rb:2479:in `open'
	from /usr/local/Homebrew/Library/Homebrew/formula.rb:2479:in `system'
	from /usr/local/Homebrew/Library/Taps/mvp/homebrew-uhubctl/Formula/uhubctl.rb:17:in `install'
	from /usr/local/Homebrew/Library/Homebrew/build.rb:177:in `block (3 levels) in install'
	from /usr/local/Homebrew/Library/Homebrew/extend/kernel.rb:500:in `with_env'
	from /usr/local/Homebrew/Library/Homebrew/build.rb:139:in `block (2 levels) in install'
	from /usr/local/Homebrew/Library/Homebrew/formula.rb:1327:in `block in brew'
	from /usr/local/Homebrew/Library/Homebrew/formula.rb:2710:in `block (2 levels) in stage'
	from /usr/local/Homebrew/Library/Homebrew/extend/kernel.rb:500:in `with_env'
	from /usr/local/Homebrew/Library/Homebrew/formula.rb:2709:in `block in stage'
	from /usr/local/Homebrew/Library/Homebrew/resource.rb:158:in `block (2 levels) in unpack'
	from /usr/local/Homebrew/Library/Homebrew/download_strategy.rb:116:in `chdir'
	from /usr/local/Homebrew/Library/Homebrew/download_strategy.rb:116:in `chdir'
	from /usr/local/Homebrew/Library/Homebrew/download_strategy.rb:103:in `stage'
	from /usr/local/Homebrew/Library/Homebrew/resource.rb:154:in `block in unpack'
	from /usr/local/Homebrew/Library/Homebrew/mktemp.rb:77:in `block in run'
	from /usr/local/Homebrew/Library/Homebrew/mktemp.rb:77:in `chdir'
	from /usr/local/Homebrew/Library/Homebrew/mktemp.rb:77:in `run'
	from /usr/local/Homebrew/Library/Homebrew/resource.rb:266:in `stage_resource'
	from /usr/local/Homebrew/Library/Homebrew/resource.rb:153:in `unpack'
	from /usr/local/Homebrew/Library/Homebrew/resource.rb:127:in `stage'
	from /usr/local/Homebrew/Library/Homebrew/vendor/portable-ruby/2.6.8_1/lib/ruby/2.6.0/forwardable.rb:230:in `stage'
	from /usr/local/Homebrew/Library/Homebrew/formula.rb:2689:in `stage'
	from /usr/local/Homebrew/Library/Homebrew/formula.rb:1320:in `brew'
	from /usr/local/Homebrew/Library/Homebrew/build.rb:133:in `block in install'
	from /usr/local/Homebrew/Library/Homebrew/extend/kernel.rb:500:in `with_env'
	from /usr/local/Homebrew/Library/Homebrew/build.rb:128:in `install'
	from /usr/local/Homebrew/Library/Homebrew/build.rb:229:in `<main>'
```

mvp, 2023-03-30:
Perhaps your libusb install is screwed up?
I have no issues, updated and reinstalled everything:

```
$ sw_vers -productVersion
11.7.4
$ brew --version
Homebrew 4.0.10-70-g6dcf028
$ brew uninstall uhubctl
$ brew uninstall libusb
$ brew update
$ brew upgrade
$ brew install libusb
$ brew tap mvp/uhubctl https://github.com/mvp/uhubctl
$ brew install libusb
$ uhubctl -v
2.5.0
```

sslupsky, 2023-03-30:
I think the problem is here:
https://github.com/mvp/uhubctl/blob/f2d97a3aadb03a2e0748f945d16a953b20915232/uhubctl.c#L32-L36

The error shows that make is using the include directory /usr/local/Cellar/libusb/1.0.26/include/libusb-1.0 and the source file specifies the include as #include <libusb-1.0/libusb.h> which resolves to /usr/local/Cellar/libusb/1.0.26/include/libusb-1.0/libusb-1.0/libusb.h which is wrong.

You do not define the include directory or search path in the make file so the wrong include directory is being used which is causing the error?

mvp, 2023-03-30:
This is really weird. It works for me on any Mac version.
One more question: does installing `pkg-config` change anything?
```
brew install pkg-config
```
Makefile will try to use `pkg-config` to properly detect libusb location (if it is available).
I do have `pkg-config` installed.

mvp, 2023-03-31:
I tried to uninstall pkg-config and uhubctl, and trying to install it again gives:
```
$ brew install uhubctl
==> Fetching dependencies for mvp/uhubctl/uhubctl: pkg-config
==> Fetching pkg-config
==> Downloading https://ghcr.io/v2/homebrew/core/pkg-config/manifests/0.29.2_3
==> Downloading https://ghcr.io/v2/homebrew/core/pkg-config/blobs/sha256:c44b1544815518726d280d92d6f6df09bd45e41ad20fd43424725c1c20760b
==> Fetching mvp/uhubctl/uhubctl
==> Downloading https://github.com/mvp/uhubctl/archive/v2.5.0.tar.gz
==> Downloading from https://codeload.github.com/mvp/uhubctl/tar.gz/refs/tags/v2.5.0
#=#=-#  #
==> Installing uhubctl from mvp/uhubctl
==> Installing dependencies for mvp/uhubctl/uhubctl: pkg-config
==> Installing mvp/uhubctl/uhubctl dependency: pkg-config
==> Pouring pkg-config--0.29.2_3.ventura.bottle.tar.gz
🍺  /usr/local/Cellar/pkg-config/0.29.2_3: 11 files, 656.1KB
==> Installing mvp/uhubctl/uhubctl
==> make
🍺  /usr/local/Cellar/uhubctl/2.5.0: 6 files, 98.9KB, built in 5 seconds
==> Running `brew cleanup uhubctl`...
Disable this behaviour by setting HOMEBREW_NO_INSTALL_CLEANUP.
Hide these hints with HOMEBREW_NO_ENV_HINTS (see `man brew`).
```

I tested this on MacOS 11.7.5 (MBP 2014) and latest 13.3 (MBP 2021 Intel), and it always works as expected.

I think your local brew or libusb setup is broken somehow.
Please try it on other Mac?

sslupsky, 2023-04-01:
Ok, I'll try it on another Mac and let you know.

I'll also try installing pkg-config ... I think that fix the make issue.

mvp, 2023-04-03:
I guess we can attribute your issue to the weirdness of your local brew install (possibly broken somehow). I have tried it on 3 different Macs - all different years and architectures, it works everywhere without any issues. Also no one else reported it except you.

Regarding supporting onboard USB C ports. The way uhubctl works is by asking libusb for all known USB hubs which advertise per port power switching, and then enumerating ports on those hubs. For whatever reason, MacBook onboard USB C ports do not appear to belong to any USB hub, at least as far as libusb concerned.

I feel that this is likely libusb bug, and it should be theoretically possible to enumerate hub that onboard USB C ports are all connected to. Or, it may be not possible - because onboard USB C ports are not even native USB, they are rather Thunderbolt 4 ports, which happen to support backwards compatibility and are able to operate in USB mode too. I can imagine that controlling power for those ports may be quite tricky - since Mac is able to also use those ports for external power charging at 100W or higher speed.

closed 2023-04-03 by mvp

sslupsky, 2023-04-03:
Yes, there was a problem with my libusb.  I am not sure why but when I looked closely in /usr/local/lib the libusb dylib's were actual files (not sim links) owned by root which caused brew to complain it could not update the links.  I suspect that whatever brew does after updating the links did not happen and that was the cause of the failure.  After I remove the files and reinstalled libusb, uhubctl correctly installed.  Apologies for the false alarm.