mvp/uhubctl #502: ExSys EX-1113HMS does not support VBUS off

mvp, 2023-04-06:
_Originally posted by @slk9 in https://github.com/mvp/uhubctl/discussions/427#discussioncomment-5533911_

I have tried ExSys EX-1113HMS today on MacOS. It does not work as expected.
Yes uhubctl will cut the connectivity so no communication from host to device connected and ports reported as off.
But all ports still provide voltage and current to devices connected. Which makes ExSys EX-1113HMS useless for most usecases when we need uhubctl.

mvp, 2023-04-06:
@slk9, thanks for reporting this. I will remove this hub from supported list

closed 2023-04-06 by mvp

F7Patrick, 2023-04-07:
Hi,

ExSys EX-1113HMS works for me under Linux with USB 2.0.
USB 3.0 doesn't work.

BR Patrick

commit 4c235868 (2023-04-06): Remove ExSys EX-1113HMS from supported list

ExSys EX-1113HMS was tested to not support turning VBUS off.
Remove from supported list.

Closes #502.