mvp/uhubctl #503: Satechi USB 3.0 Typ C Slim Hub (1x HDMI, 2x USB-A, 1x USB-C) working with VBUS switched too

hermannkm, 2023-04-07:
```
Current status for hub 0-2 [2109:0813 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 20-8 [2109:2813 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 1: 0103 power enable connect [...]
  Port 2: 0103 power enable connect [...]
  Port 3: 0503 power highspeed enable connect [2109:d101 VIA Labs, Inc. USB Keyboard 0000000000000001]
  Port 4: 0100 power

```
Working for my audio interface and microphone which did not sleep while my iMac did.

hermannkm, 2023-04-07:
Just tested. VBUS not switched off.

mvp, 2023-04-09:
Thanks for reporting. These days majority of new hubs advertise power switching, but not support vbus off. Not adding to the table.

closed 2023-04-09 by mvp