mvp/uhubctl #504: lsusb reports power switching supported, uhubctl says not

stefan-sherwood, 2023-04-12:
I am using a 4-port USB 3.2 Gen1 hub made by SmartQ Inc. plugged into a USB 3 port on a Raspberry Pi 4B running Raspberry Pi OS 11 (bullseye). I have enabled permissions by copying https://github.com/mvp/uhubctl/blob/master/udev/rules.d/52-usb.rules unmodified into `/etc/udev/rules.d/52-usb.rules` and rebooting.

lsusb reports it supports per-port power switching:
```
$ lsusb -v -D /dev/bus/usb/002/003

Device: ID 0bda:0411 Realtek Semiconductor Corp. Hub
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.20
  bDeviceClass            9 Hub
  bDeviceSubClass         0
  bDeviceProtocol         3
  bMaxPacketSize0         9
  idVendor           0x0bda Realtek Semiconductor Corp.
  idProduct          0x0411 Hub
  bcdDevice            1.01
  iManufacturer           1 Generic
  iProduct                2 USB3.2 Hub
  iSerial                 0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x001f
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower                0mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0
      bInterfaceProtocol      0 Full speed (or root) hub
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes           19
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Feedback
        wMaxPacketSize     0x0002  1x 2 bytes
        bInterval               8
        bMaxBurst               0
Hub Descriptor:
  bLength              12
  bDescriptorType      42
  nNbrPorts             4
  wHubCharacteristic 0x0009
    Per-port power switching
    Per-port overcurrent protection
  bPwrOn2PwrGood       50 * 2 milli seconds
  bHubContrCurrent      8 milli Ampere
  bHubDecLat          0.2 micro seconds
  wHubDelay          3202 nano seconds
  DeviceRemovable    0x00
 Hub Port Status:
   Port 1: 0000.02a0 lowspeed L1
   Port 2: 0000.02a0 lowspeed L1
   Port 3: 0000.02a0 lowspeed L1
   Port 4: 0000.02a0 lowspeed L1
Binary Object Store Descriptor:
  bLength                 5
  bDescriptorType        15
  wTotalLength       0x002a
  bNumDeviceCaps          3
  USB 2.0 Extension Device Capability:
    bLength                 7
    bDescriptorType        16
    bDevCapabilityType      2
    bmAttributes   0x0000f41e
      BESL Link Power Management (LPM) Supported
    BESL value     1024 us
    Deep BESL value    61440 us
  SuperSpeed USB Device Capability:
    bLength                10
    bDescriptorType        16
    bDevCapabilityType      3
    bmAttributes         0x00
    wSpeedsSupported   0x000e
      Device can operate at Full Speed (12Mbps)
      Device can operate at High Speed (480Mbps)
      Device can operate at SuperSpeed (5Gbps)
    bFunctionalitySupport   1
      Lowest fully-functional device speed is Full Speed (12Mbps)
    bU1DevExitLat          10 micro seconds
    bU2DevExitLat        1023 micro seconds
  Container ID Device Capability:
    bLength                20
    bDescriptorType        16
    bDevCapabilityType      4
    bReserved               0
    ContainerID             {20b9cde5-7039-e011-a935-0002a5d5c51b}
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x000d
  Self Powered
  U1 Enabled
  U2 Enabled
```

The device is listed as hub 2-1:
```
$ sudo uhubctl

Current status for hub 2-1 [0bda:0411 Generic USB3.2 Hub, USB 3.20, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 2 [1d6b:0003 Linux 6.1.21-v8+ xhci-hcd xHCI Host Controller 0000:01:00.0, USB 3.00, 4 ports, ppps]
  Port 1: 0263 power 5gbps U3 enable connect [0bda:0411 Generic USB3.2 Hub, USB 3.20, 4 ports, ppps]
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 1-1.1 [0bda:5411 Generic USB2.1 Hub, USB 2.10, 5 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
  Port 5: 0100 power
Current status for hub 1-1 [2109:3431 USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 1: 0507 power highspeed suspend enable connect [0bda:5411 Generic USB2.1 Hub, USB 2.10, 5 ports, ppps]
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
Current status for hub 1 [1d6b:0002 Linux 6.1.21-v8+ xhci-hcd xHCI Host Controller 0000:01:00.0, USB 2.00, 1 ports, ppps]
  Port 1: 0507 power highspeed suspend enable connect [2109:3431 USB2.0 Hub, USB 2.10, 4 ports, ppps]
```

Yet when I try to turn the ports off I get this error:
```
$ sudo uhubctl -l 2-1 -a off
No compatible devices detected at location 2-1!
Run with -h to get usage info.
```

Am I missing something?

mvp, 2023-04-30:
Closing for inactivity. @stefan-sherwood, please reopen if uhubctl version 2.4.0 or later doesn't work with Raspberry Pi 4B.

closed 2023-04-30 by mvp

stefan-sherwood, 2023-04-30:
I updated and compiled to the latest version of uhubctl.

$ sudo ./uhubctl -v
2.5.0-12-g4c235868

The behavior hasn't changed: `lsusb` reports per-port power switching, `uhubctl` reports that it's not supported, identical messages to previous report.

stefan-sherwood, 2023-04-30:
I can't reopen this report since I'm not the one who closed it.

stefan-sherwood, 2023-05-02:
1. (use different hub) This looks like it's going to have to be the answer
2. I tried controlling the ports overriding the duality matching as per your instruction and it gives no error but also doesn't actually turn off the port (it shows the ports as "off" but I have a mini light plugged in to one of the ports and it's still on). From reading your doc that suggests the hub may not actually support VBUS off.
3. I could make a patch myself but given the failed attempt using the `-e` switch, I don't think it will work.

I guess I'm back to the drawing board searching for a modern high speed USB hub that supports per-port power switching.

Thanks for your help.

mvp, 2023-05-03:
For RPi4B I highly recommend getting UUGear Mega4 - it was specifically designed to work with RPi4 and to support `uhubctl` by design. It supports stacking too, all within RPi4 physical contstraints. And it is not that expensive.

SlySven, 2023-05-03:
Might I suggest you investigate the UUGear [Mega4](https://www.uugear.com/product/mega4-4-port-usb-3-ppps-hub-for-raspberry-pi-4b/) product - they have worked with this project to ensure it works on their produce - as you can see by visiting #316 .

Disclosure: I and @mvp both received a free sample of this unit! :innocent: