mvp/uhubctl #506: UUGear MEGA4 (for Raspberry Pi 4B) is detected as 2 hubs when using USB-C connection

AubinF, 2023-04-27:
Hi,

Great library :)

I have a small issue with UUGear MEGA4 (for Raspberry Pi 4B) on Ubuntu 22 LTS though.

Until now I have been using the hub with a USB3 male to female adapter (similar to [this one](https://m.media-amazon.com/images/I/71aPOaDZ44L._AC_SL1500_.jpg)). The output of uhubctl is the following:

> $ uhubctl 
> Current status for hub 1-3 [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
>   Port 1: 0103 power enable connect [054c:0994 Sony USB Charger]
>   Port 2: 0100 power
>   Port 3: 0100 power
>   Port 4: 0100 power

Now I have to use a different machine which has only USB-C sockets, so I tried a USB-C male to female adapter (similar to [this one](https://ae01.alicdn.com/kf/Sbfd69ca907704a4499f77b11fbd9e659M/Cabletime-USB-C-OTG-Cable-Type-C-Male-To-USB-Female-Adapter-5GBPS-Transmission-Fast-Charge.jpg)). In this case the output of uhubctl is the following:

> $ uhubctl 
> Current status for hub 4-1 [2109:0817 VIA Labs, Inc. USB3.0 Hub, USB 3.10, 4 ports, ppps]
>   Port 1: 02a0 power 5gbps Rx.Detect
>   Port 2: 02a0 power 5gbps Rx.Detect
>   Port 3: 02a0 power 5gbps Rx.Detect
>   Port 4: 02a0 power 5gbps Rx.Detect
> Current status for hub 1-4 [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
>   Port 1: 0103 power enable connect []
>   Port 2: 0100 power
>   Port 3: 0100 power
>   Port 4: 0100 power

Not sure if this is the normal behaviour. I would expect the listing to be the same regardless of the type of cable I use to connect the hub to the computer. I can patch my code to work in any case but I would prefer if this was handled on uhubctl's side.

Thanks

mvp, 2023-04-27:
Yes, this is expected behavior. Second is showing USB 3.0 connection that becomes available when you use USB 3 compliant cable and upstream hub. This is covered in readme, look for USB 3 duality.

closed 2023-04-27 by mvp

AubinF, 2023-04-27:
Hi, thank you for your quick reply. Will definitely have a look at the readme. thanks!