mvp/uhubctl #509: StarTech hub with different USB ID doesn't work

mkohler, 2023-05-16:
I just bought a StarTech HB30A4AIB, which is a model listed on the README. However, this hub has USB ID `14b0:01ef`, which is different than the ID of the HB30A4AIB in the README. I tested this hub, and while it does report that it turned off the power on a port, it doesn't actually turn off the power.

mvp, 2023-05-17:
That's pretty funny - break per port power switching to improve USB-IF compliance 🤣.
I will add a readme note to this issue tomorrow, thanks for reporting!

closed 2023-05-23 by mvp

daler-rahimov, 2024-10-11:
@mvp you have a wrong VID:PID in the README file for the warning, `2109:2817` instead of `14b0:01ef`

mvp, 2024-10-12:
This is quite common situation. USB vendor 2109 is Via Labs, Inc - company that manufactures USB chipsets. 14b0 is Startech Inc (hub manufacturer). Many hub manufacturers start by using USB chipset unconfigured and by default having identity of chipset maker. Over time, they may pay small fee to USB IF to obtain their custom vendor id and configure USB chipset to carry that id.

colorimeter, 2026-09-01:
Maybe a firmware update resolves the issue https://wiki.odroid.com/odroid-c4/software/vl817_usb-hub_fw_update

commit 1d3b8414 (2023-05-23): Add a warning link for StarTech HB30A4AIB different batch

Add a warning link for StarTech HB30A4AIB - new hardware is using
different USB chipset and different USB vid:pid,
which does not support power switching correctly.

Closes #509.