mvp/uhubctl #510: Permission problem on Pi4b, Raspberry Pi OS

kbasaran, 2023-05-25:
### Discussed in https://github.com/mvp/uhubctl/discussions/508

<div type='discussions-op-text'>

<sup>Originally posted by **kbasaran** May  6, 2023</sup>
Hello all,
I couldn't get to run uhubctl without sudoing. What am I missing, can anybody help please?
I have copied the 52-usb.rules file under /etc/udev/rules.d/ and made no changes on it except for removing the line for Linux version <6.
```
pi@amp2023:/etc/udev/rules.d $ groups
pi adm dialout cdrom sudo audio video plugdev games users input render netdev lpadmin gpio i2c spi

pi@amp2023:/etc/udev/rules.d $ uname -a
Linux amp2023 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux

pi@amp2023:/etc/udev/rules.d $ uhubctl --version
2.5.0-13-g2f45ac61

pi@amp2023:/etc/udev/rules.d $ uhubctl
There were permission problems while accessing USB.
Follow https://git.io/JIB2Z for a fix!
No compatible devices detected!
Run with -h to get usage info.

pi@amp2023:/etc/udev/rules.d $ sudo uhubctl
Current status for hub 2 [1d6b:0003 Linux 6.1.21-v8+ xhci-hcd xHCI Host Controller 0000:01:00.0, USB 3.00, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 1-1 [2109:3431 USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0503 power highspeed enable connect [1235:8213 Focusrite Scarlett 8i6 USB F8GZU561B0C88D]
  Port 3: 0100 power
  Port 4: 0100 power
Current status for hub 1 [1d6b:0002 Linux 6.1.21-v8+ xhci-hcd xHCI Host Controller 0000:01:00.0, USB 2.00, 1 ports, ppps]
  Port 1: 0503 power highspeed enable connect [2109:3431 USB2.0 Hub, USB 2.10, 4 ports, ppps]

pi@amp2023:/etc/udev/rules.d $ sudo rpi-eeprom-update
*** UPDATE AVAILABLE ***
BOOTLOADER: update available
   CURRENT: Thu  3 Sep 12:11:43 UTC 2020 (1599135103)
    LATEST: Wed 11 Jan 17:40:52 UTC 2023 (1673458852)
   RELEASE: default (/lib/firmware/raspberrypi/bootloader/default)
            Use raspi-config to change the release.

  VL805_FW: Dedicated VL805 EEPROM
     VL805: up to date
   CURRENT: 000138c0
    LATEST: 000138c0
```
I did a restart already.
Device is Pi4b v1.2, running Pi OS.</div>

mvp, 2023-06-03:
Can you try updating bootloader to most recent version?

CasperOng, 2023-06-08:
Did you follow the [tutorial for raspberry pi](https://github.com/mvp/uhubctl#linux-usb-permissions)?
In particular, for Raspberry Pi 4B, you need to add these 2 lines to file `/etc/udev/rules.d/52-usb.rules`:
```SUBSYSTEM=="usb", ATTR{idVendor}=="2109", MODE="0666"
SUBSYSTEM=="usb", ATTR{idVendor}=="1d6b", MODE="0666"```
then either reboot or run this command to apply changes live:
```sudo udevadm trigger --attr-match=subsystem=usb```

and this worked for me, try to run it without sudo

kbasaran, 2023-06-14:
> Did you follow the [tutorial for raspberry pi](https://github.com/mvp/uhubctl#linux-usb-permissions)? In particular, for Raspberry Pi 4B, you need to add these 2 lines to file [...]

I did all this yes. This is the contents of my 52-usb.rules file:
```
SUBSYSTEM=="usb", DRIVER=="usb", MODE="0666", ATTR{idVendor}=="2109" \
SUBSYSTEM=="usb", DRIVER=="usb", MODE="0666", ATTR{idVendor}=="1d6b" \
  RUN+="/bin/sh -c \"chown -f root:dialout $sys$devpath/*-port*/disable || true\"" \
  RUN+="/bin/sh -c \"chmod -f 660 $sys$devpath/*-port*/disable || true\""
```

mvp, 2023-06-19:
Unfortunately, I don't have raspberry pi 4 to test. I recommend debugging this with gdb. It is quite simple:
```
make
gdb ./uhubctl
set args <command line params>
b <line number>
c
p <variable>
l  <list source>
```
You would need to debug location that changes power state.

mvp, 2023-07-05:
@kbasaran, can you try disabling Linux 6 specific code? To do this, you need to add older udev rules (that y removed), and use `uhubctl -S` to control power. In general, you should keep both old and new rules present, it does no harm to have both.

kbasaran, 2023-07-05:
OK, that solves it. Keeping only this section
```
# This is for Linux before 6.0:
SUBSYSTEM=="usb", DRIVER=="usb", MODE="0664", GROUP="dialout"
```
or this section together with Linux 6 specific section works. Thank you very much.

mvp, 2023-07-07:
That must mean you actually had Linux kernel 5. Oh well, good news it works for you now!

closed 2023-07-07 by mvp

kbasaran, 2023-07-07:
Thank you. In my first post above you can see that I get the response 6.1.21 when I run "uname -a". I had based myself on that. I may surely be mistaken though.