mvp/uhubctl #512: Dell Wyse 3040

jsv93, 2023-07-12:
After testing in a debian environment per port power control was fully functional. -f was required however as the device lists the hubs as not having power management options.

Currently using this to switch a stream deck off overnote on a satellite bitfocus companion node.

mvp, 2023-07-15:
How are you testing that power control was functional? Did you test that voltage actually was turned off?

mvp, 2023-07-15:
If hub didn't advertise power switching support, I highly doubt it will support turning voltage off. I recommend testing it by connecting a phone, turning port off and checking if phone stops charging.

jsv93, 2023-07-15:
It definately works. I tested using a phone to see if it was charging or not and every port works. Plus it has been working everyday since. I have an automation in Home assistant to turn off the stream deck but not the devices associated with my hyperHDR setup and it works flawlessly.

mvp, 2023-07-18:
Can you please provide `uhubctl -f` output?
I need this to properly fill in in device table, in particular number of USB ports, whether they are USB3 or USB2 and so on.
Thanks!

jsv93, 2023-07-18:
Here is the output from `uhubctl -f`

> Current status for hub 2 [1d6b:0003 Linux 5.10.0-23-amd64 xhci-hcd xHCI Host Controller 0000:00:14.0, USB 3.00, 6 ports, nops]
>   Port 1: 02a0 power 5gbps Rx.Detect
>   Port 2: 02a0 power 5gbps Rx.Detect
>   Port 3: 02a0 power 5gbps Rx.Detect
>   Port 4: 0080 off
>   Port 5: 02a0 power 5gbps Rx.Detect
>   Port 6: 02a0 power 5gbps Rx.Detect
> Current status for hub 1 [1d6b:0002 Linux 5.10.0-23-amd64 xhci-hcd xHCI Host Controller 0000:00:14.0, USB 2.00, 7 ports, nops]
>   Port 1: 0100 power
>   Port 2: 0100 power
>   Port 3: 0100 power
>   Port 4: 0000 off
>   Port 5: 0100 power
>   Port 6: 0100 power
>   Port 7: 0100 power

closed 2023-07-30 by mvp

kkubicki, 2024-02-10:
In my case - no luck, VBUS stays on and charging remains (although enumerated devices are disconnected and reconnected). What is the full command that you use to toggle the power?
My hub status is the same apart from newer kernel (6.6.13).

killerbees19, 2025-08-25:
I had no luck either, doesn't work on two Dell Wyse 3040 with `-f`.

Maybe this entry should be removed from the compatibility list?

steve-m, 2026-10-30:
I also tested it on my Wyse 3040, and while the devices disappear from the bus, the VBUS to the devices (tried all ports) is constant on and cannot be powered off.

commit 54ab9e72 (2023-07-30): Add Dell Wyse 3040 as compatible device

Dell Wyse 3040 is odd - it has 6 ports advertised on USB2 and 7 ports on USB3,
and both USB2+USB3 do not advertise power switching support.
However, using -f it seems that power switching does work.
I assume it needs separate control for USB2+USB3 using -e as well.

Closes #512.