mvp/uhubctl #514: new supported hub

bm16ton, 2023-07-14:
coolgear.com model USBG-12U2ML 12port usb2.0 hub. Still for sale tho I dunno if just selling old stock or actively producing. The datasheet for V2 is dated 2022, So I must  have a V1, the oldest review posted was from 2012. The datasheet is here https://www.coolgear.com/wp-content/uploads/2013/05/USBG-12U2ML-Product-Manual-v2.pdf Funny the usb pid's/vid's are associated with "Genesys Logic, Inc. Logitech G110 Hub" product description. I realize lsusb uses its own off device databse for pid/vid to Manufacturer/model just found it amusing. Great hub if not over priced, used a usb-led for port power control test. Awesome software thank you very much for writing it! Heres the uhubctl info  PS the writeup said to look for "Per-port power switching" as confirmation of support, mine doesnt appear to use those words instead the abreviated "ppps" seems worth noting.

 Current status for hub 3-1.1.2.4.3 [05e3:0607 USB2.0 Hub, USB 2.00, 4 ports, ppps]
  Port 1: 0103 power enable connect [046d:c52b Logitech USB Receiver]
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
Current status for hub 3-1.1.2.4.2 [05e3:0607 USB2.0 Hub, USB 2.00, 4 ports, ppps]
  Port 1: 0103 power enable connect [1532:0053 Razer Razer Naga Chroma]
  Port 2: 0100 power
  Port 3: 0103 power enable connect [320f:5044 Glorious GMMK Pro ANSI vial:f64c2b3c]
  Port 4: 0503 power highspeed enable connect [13fd:1840 Generic External 57442D574343334637545652]
Current status for hub 3-1.1.2.4.1 [05e3:0607 USB2.0 Hub, USB 2.00, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0103 power enable connect [067b:2303 Prolific Technology Inc. USB-Serial Controller]
Current status for hub 3-1.1.2.4 [05e3:0607 USB2.0 Hub, USB 2.00, 4 ports, ppps]
  Port 1: 0503 power highspeed enable connect [05e3:0607 USB2.0 Hub, USB 2.00, 4 ports, ppps]
  Port 2: 0503 power highspeed enable connect [05e3:0607 USB2.0 Hub, USB 2.00, 4 ports, ppps]
  Port 3: 0503 power highspeed enable connect [05e3:0607 USB2.0 Hub, USB 2.00, 4 ports, ppps]
  Port 4: 0100 power

And did you want the usb hubs lsusb display of decriptors? Its a lot for all 4 usb devices but here it is.

Bus 003 Device 020: ID 05e3:0607 Genesys Logic, Inc. Logitech G110 Hub
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            9 Hub
  bDeviceSubClass         0 
  bDeviceProtocol         2 TT per port
  bMaxPacketSize0        64
  idVendor           0x05e3 Genesys Logic, Inc.
  idProduct          0x0607 Logitech G110 Hub
  bcdDevice            8.05
  iManufacturer           0 
  iProduct                1 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0029
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Single TT
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
[... 205 lines cut ...]
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0001  1x 1 bytes
        bInterval              12
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       1
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0 
      bInterfaceProtocol      2 TT per port
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0001  1x 1 bytes
        bInterval              12

closed 2023-07-18 by mvp

mvp, 2023-07-18:
Thank you for your report, added this hub to supported list.
Couple notes:
1. It seems that this hub topology is well built, it uses 4 x 4port USB chips, with one acting as root for 3 others. It also means 1 port was not wired outside - it could have been 13 port hub if Coolgear wanted it.
2. Your topology shows that hub is connected too deep at 3-1.1.2.4. Ideally, it should be much shorter, e.g. 3-1 or 3-1.1. The longer USB chain is, the worse it will operate. And, since USB has limit to USB device chain at 7, it means you most likely not be able to attach any other hub as downstream to it. I recommend attaching this hub as close to root bus as possible, ideally plug it directly to your root hub - everything will work faster and USB enumeration should be quicker.

bm16ton, 2023-07-18:
Wow spot on! I have  it plugged into a root port (usb2 root port but the unused usb3 is not root that got me once) and the other day I pplugged in a device that enumerates as a hub with multiple devices and linux wouldn't enumerate as to many deep. Couldn't for the life of me figure it out. Took me a bit, but took you no time at all :) if I had half your skills! Amazing project! Thank you for it! 
ps they make this hub as usb3 as well but both the usb3 and usb3 versions aare expensive alas all industrial stuff and sorry I couldn't nail down a release date.

commit b16ba1b3 (2023-07-18): Add Coolgear USBG-12U2ML as compatible device

Closes #514.