mvp/uhubctl #516: CH335F is a fully compatible USB hub chip

DeqingSun, 2023-07-16:
I just tested the CH335F USB hub chip from WCH and it is fully compatible with uhubctl. The chip can be configured as individual control or gang control, and it can generate the gpio signal to cut or resume voltage.

This chip is fairly new and I have not found an off-the-shelf product with that chip. And I know that an average user is not capable of using that chip. However, that chip would be useful for an average electronic engineer to implement their own hub. Compared to other chips such as FE2.1 or USB2514B, the CH335F is more compact with lower pin count and price. And the switch/current sensing combo CH217K has an adjustable current threshold.

For anyone interested, I also have open sourced design for my own usage. https://oshwlab.com/iamseer/ch335-hub

Disclaimer: I'm not affiliated with the WCH, just an enthusiastic poking around different chips.

mvp, 2023-07-17:
To the best of my knowledge, almost all USB chipsets do support per port power switching (e.g. Realtek, Via, TI, Renesas, etc). However, most hub manufacturers don't configure ppps properly, or simply don't provide proper circuitry to make switching actually work.

Also, purpose of the table of supported devices is to provide a list of off the shelf USB hubs which are easy to purchase, and which properly support ppps without a need to solder or tweak anything.

Unfortunately, your USB chip doesn't really meet these objectives, so we can't add a chip to the table.

That said, if you can find commercially available USB hub which utilitizes this chip and supports ppps correctly, I would be happy to add it to the list.

closed 2023-07-17 by mvp

DeqingSun, 2023-07-17:
@SlySven 

The CH335F come with VID/PID: 1a86 8094, if you do not use an optional external EEPROM. But if you attach an EEPROM, you can configure the PID/VID, product string, and some misc configuration. Either way the firmware is already in the hub chip.

mvp, 2023-07-18:
While I don't think this belongs to supported devices table (as not easily commercially available solution), it is definitely worth adding to notable project list for hardware enthusiasts like @SlySven :-) - I'm going to do that shortly.
Thank you @DeqingSun for your open sourced hardware design! 👍

mvp, 2023-07-18:
@DeqingSun, added a link to your site https://oshwlab.com/iamseer/ch335-hub to README. Thanks again!

DeqingSun, 2023-07-18:
@mvp Thanks for adding the project. Also you comment "almost all USB chipsets do support per port power switching" changed my impression about this feature. During my research I found most USB chipsets that has pins on 4 sides (QFP, QFN) generally supports PPPS. While none of the chip that has pins on 2 sides (SOP, SSOP) has PPPS due to low pin count. However, the low pin count ones are more common due to lower cost and easier assembly. That's why I used to think most hub chip does not support PPPS.

chuckleu1976, 2023-09-06:
Hi!
I designed a new hardware by using CH335F, and tried to use it to control USB power.  But I find uhubctl does not support it. 
Here is the message!
[image]
[image]

DeqingSun, 2023-09-06:
@chuckleu1976 you would either need to use EEPROM or you need to make sure PGANG is pulled down during boot. Or it will be default gang control.

DeqingSun, 2023-09-06:
@chuckleu1976 That is different from my output.
[image]

chuckleu1976, 2023-09-07:
Thank you!  I have checked your output.  You connect the CH335F to a Raspberry PI USB port.  I connect all 4 CH335Fs to a USB hub chip FS2.1s , then through FS2.1s connect to Raspberry PI. I think it should work. But I don't know why uhubctl can't recognize it.
Do you have some good ideas to make it work?
Or you have done something to configure the CH335F chipset? 

> @chuckleu1976 That is different from my output. [image]

chuckleu1976, 2023-09-07:
Hi, Deqing!  I left the No.1 pin and No.3 pi unconnected, and the chip is configured as ganded mode. So I need to pull the No3 pins to ground. Is it correct ?

chuckleu1976, 2023-09-07:
Thanks, I have changed my design and made it just like what you said.

DeqingSun, 2023-09-07:
For anyone got a similar issue:

You would either need to use EEPROM or you need to make sure PGANG is pulled down during boot. Or it will be default gang control.

Here is how I did it with 9 LED configuration. In boot up, the LED3 will be LOW and LED4 will be HIGH. An resistor with correct resistance will pull the voltage on LED1 and LED2 (internal pulled up) to LOW level. And that resistor shouldn't be too small or there will be significant LED ghosting in scanning. 

[image]

chuckleu1976, 2023-09-11:
Hi, I have redesigned the hardware. Here is the output. I think It supports ppps.
 
[image]
[image]

mvp, 2023-09-22:
@chuckleu1976, if you are offering this hardware for sale to public, I would be happy to add it to supported table. Basic requirement to be added to supported table is ability for average consumer to buy it somewhere.

commit 30372ce5 (2023-07-18): Add open source hardware hub design based on CH335F to notable projects

Closes #516.