mvp/uhubctl #520: Aukey Cb-h15 Superspeed 3-port Usb 3.0 Hub With 10/100/1000 Gigabit Ethernet

aisamu, 2023-09-06:
Reporting a new device!

Unfortunately, there's no VBUS control :(

```
Current status for hub 1-1.1 [2109:2812 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 1: 0303 power lowspeed enable connect [0101:0007 USB OPTICAL MOUSE]
  Port 2: 0303 power lowspeed enable connect [04d9:1400]
  Port 3: 0303 power lowspeed enable connect [1bcf:0005 USB Optical Mouse]
  Port 4: 0100 power
Current status for hub 1-2.1 [2109:0812 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 0203 power 5gbps U0 enable connect [0bda:8153 Realtek USB 10/100/1000 LAN F01E341F851F]
```

Thanks a lot for this!

aisamu, 2023-09-06:
```Current status for hub 1-1.1 [2109:2812 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 3: 0303 power lowspeed enable connect [1bcf:0005 USB Optical Mouse]
Sent power off request
New status for hub 1-1.1 [2109:2812 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 3: 0000 off
Current status for hub 1-2.1 [2109:0812 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports, ppps]
  Port 3: 02a0 power 5gbps Rx.Detect
Sent power off request
New status for hub 1-2.1 [2109:0812 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports, ppps]
  Port 3: 0080 off```
  
  Both were turned off and the pesky decorative light on the side of the mouse was still on

mvp, 2023-09-09:
Sorry, but we only add devices which actually support vbus off. If you can find one that works correctly, please report! Thanks.

closed 2023-09-09 by mvp