mvp/uhubctl #525: Supported hub: Sugoi HUB by System Talks (model: USB2-HUB4X-BK)

ChrisVilches, 2023-09-18:
I had this issue so I posted on StackOverflow/askubuntu (I just post this for background):

https://askubuntu.com/questions/1486016/how-can-i-turn-off-a-usb-lamp-in-ubuntu?noredirect=1#comment2602649_1486016

But I solved it myself by buying this hub:

https://www.system-talks.co.jp/product/sgc-4X.htm

As I explain in the question, the lamp I used is just a dumb device that receives energy, which means the power-off is a true power-off and not just a logical disconnect. I only tested one port.

ChrisVilches, 2023-09-19:
@mvp I was under the impression that being able to turn ON/OFF the lamp was enough to prove PPPS worked correctly. What else should I test?

mvp, 2023-09-19:
Yes, turning lamp off proves vbus test is working. I am only asking about other ports being able to turn power on/off independently. I have seen some hubs which turn it off for all ports (ganged), and I want to add this device to compatible table we appropriate information. Thanks!

ChrisVilches, 2023-09-21:
@mvp Yes, it only turns off/on the selected ports.

I have a Pocket Wi-Fi connected and it remains ON when I turn off the lamp.

mvp, 2023-09-22:
Thank you @ChrisVilches, added this hub to compatible list.

commit 5b2c40db (2023-09-22): Add Sugoi USB2-HUB4X as compatible device

Closes #525.
closed 2023-09-22 by mvp