mvp/uhubctl #526: Switch port on active hub

sralus, 2023-09-22:
Hello
I have a Raspberry 4 with several USB devices.
I have an Activeb USB3 HUB on the lower USB 3.0 port.
`https://www.amazon.de/CSL-Verteiler-Ultrabook-Tablet-PC-bus-powered/dp/B01K7RR3W8/ref=asc_df_B01K7RR3W8/?tag=googshopde-21&linkCode=df0&hvadid=308897333482&hvpos=&hvnetw=g&hvrand=9424246316756416189&hvpone=&hvptwo=&hvqmt=&hvdev=c&hvdvcmdl=&hvlocint=&hvlocphy=9043746&hvtargid=pla-562378000734&psc=1&th=1&psc=1&tag=&ref=&adgrpid=61522440175&hvpone=&hvptwo=&hvadid=308897333482&hvpos=&hvnetw=g&hvrand=9424246316756416189&hvqmt=&hvdev=c&hvdvcmdl=&hvlocint=&hvlocphy=9043746&hvtargid=pla-562378000734`
I connected a ESP32 to it.
This connects to my JK-BMS (battery management) via Bluetooth.
Every now and then I have to restart the ESP32 because the Bluetooth connection breaks down.
If I could now briefly switch the USB port on/off using uhubctl, that would be great!!
Unfortunately, I don't know how to access the USB port of the ESP32 directly.
The device is this one:
 Port 2: 0103 power enable connect [10c4:ea60 Silicon Labs CP2102 USB to UART Bridge Controller 0001]

`Current status for hub 2-2 [0bda:0411 Generic USB3.2 Hub, USB 3.20, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 0203 power 5gbps U0 enable connect [0bda:9210 ULT-Best Best USB Device 012938000FA5]
Current status for hub 2 [1d6b:0003 Linux 6.1.21-v8+ xhci-hcd xHCI Host Controller 0000:01:00.0, USB 3.00, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 0203 power 5gbps U0 enable connect [0bda:0411 Generic USB3.2 Hub, USB 3.20, 4 ports, ppps]
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 1-1.2 [0bda:5411 Generic USB2.1 Hub, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0103 power enable connect [10c4:ea60 Silicon Labs CP2102 USB to UART Bridge Controller 0001]
  Port 3: 0103 power enable connect [0403:6001 FTDI FT232R USB UART B0004YM7]
  Port 4: 0100 power
Current status for hub 1-1 [2109:3431 USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0503 power highspeed enable connect [0bda:5411 Generic USB2.1 Hub, USB 2.10, 4 ports, ppps]
  Port 3: 0303 power lowspeed enable connect [0665:5161]
  Port 4: 0303 power lowspeed enable connect [0665:5161]
Current status for hub 1 [1d6b:0002 Linux 6.1.21-v8+ xhci-hcd xHCI Host Controller 0000:01:00.0, USB 2.00, 1 ports, ppps]
  Port 1: 0503 power highspeed enable connect [2109:3431 USB2.0 Hub, USB 2.10, 4 ports, ppps]
`
` sudo lsusb -v

Bus 002 Device 003: ID 0bda:9210 Realtek Semiconductor Corp. RTL9210 M.2 NVME Adapter
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.20
  bDeviceClass            0
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0         9
  idVendor           0x0bda Realtek Semiconductor Corp.
  idProduct          0x9210 RTL9210 M.2 NVME Adapter
  bcdDevice           20.01
  iManufacturer           1 ULT-Best
  iProduct                2 Best USB Device
  iSerial                 3 012938000FA5
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0079
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              896mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         8 Mass Storage
[... 822 lines cut ...]
        wMaxPacketSize     0x0004  1x 4 bytes
        bInterval              12
Hub Descriptor:
  bLength               9
  bDescriptorType      41
  nNbrPorts             1
  wHubCharacteristic 0x0009
    Per-port power switching
    Per-port overcurrent protection
    TT think time 8 FS bits
  bPwrOn2PwrGood       10 * 2 milli seconds
  bHubContrCurrent      0 milli Ampere
  DeviceRemovable    0x02
  PortPwrCtrlMask    0xff
 Hub Port Status:
   Port 1: 0000.0503 highspeed power enable connect
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0001
  Self Powered
`

Which command should I enter for this?

Thank you in advance

mvp, 2023-09-22:
This should work as long as your hub supports power switching (not a given):
```
uhubctl -l 1-1.2 -p 2 -a 0
```
Please read documentation, it should have all quirks explained.

closed 2023-09-22 by mvp

sralus, 2023-09-22:
Thank you for the info!
I sent the comand and got this output:

`sudo uhubctl -l 1-1.2 -p 2 -a 0
Current status for hub 2-2 [0bda:0411 Generic USB3.2 Hub, USB 3.20, 4 ports, ppps]
  Port 2: 02a0 power 5gbps Rx.Detect
Sent power off request
New status for hub 2-2 [0bda:0411 Generic USB3.2 Hub, USB 3.20, 4 ports, ppps]
  Port 2: 00a0 off
Current status for hub 1-1.2 [0bda:5411 Generic USB2.1 Hub, USB 2.10, 4 ports, ppps]
  Port 2: 0103 power enable connect [10c4:ea60 Silicon Labs CP2102 USB to UART Bridge Controller 0001]
Sent power off request
New status for hub 1-1.2 [0bda:5411 Generic USB2.1 Hub, USB 2.10, 4 ports, ppps]
  Port 2: 0000 off`

Unfortunately the USB connection remains active, so the ESP32 on the USB port does not restart. Aslo, my USB hub does not support power switching.
Is it possible to switch only one USB 3.0 port directly on the Raspberry?
Thank you in advance

mvp, 2023-09-23:
I'm not sure why it doesn't work for you. Chances are that your hub doesn't support turning off power at the hardware level.

Unfortunately, Raspberry Pi hardware only supports turning USB ports off and on all at once, not individually. As explained in FAQ, to solve this, you can buy any USB hub listed in the table or supported devices - all of them should work.

If you are interested in specifically Raspberry Pi 4B, there is an excellent option: UUGear Mega4. It was designed for RPi4B to specifically support uhubctl operation on all 4 ports individually.

I have Mega4 in my possession, and can attest it works perfectly.