mvp/uhubctl #527: StarTech.com 5G4AIBS-USB-HUB-NA 4 port hub works with uhubctl

robertrau, 2023-09-25:
I just received a StarTech.com 5G4AIBS-USB-HUB-NA 4 port hub from Amazon.
It reports power control:
**$ ./uhubctl
Current status for hub 20-1 [14b0:01d3 VIA Labs, Inc. USB2.0 Hub 000000000, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
Current status for hub 1-1 [14b0:01d2 VIA Labs, Inc. USB3.0 Hub 000000000, USB 3.20, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect**
I tested it with devices in 3 of the 4 sockets and was able to control each port, but with a odd order.
Starting from the power connector end, ports are numbered 3, 4, 1, 2
Each port has a small blue LED indicating it power state.
Each USB socket also has a manual switch. If the switch is off, the port is always off. If the switch is on, you have power control.

Power control command was just:
./uhubctl -a off -p 3 -l 20-1
Tested on a MacBook running MacOS 10.13.6
StarTech.com also has a similar 7 port hub that they claim also has per-port power control.

mvp, 2023-09-26:
Thanks for your report! Can you please check vbus off support? That is, plug USB fan or light, or at least your phone, turn power off to that port and confirm that it is not supplying any voltage anymore?

robertrau, 2023-09-28:
mvp,

Sorry, my error. I was testing with a USB power meter. Apparently its backlight has a timeout, after getting things setup, port turned off, the backlight shut off. I am now testing with just LED lights and it doesn’t switch Vusb as I thought it did.

Bob

mvp, 2023-10-01:
Thats unfortunate :cry:.
I will close this issue then.
Please report any other hub if it supports vbus off.
Thank you!

closed 2023-10-01 by mvp