mvp/uhubctl #528: Hubs No Longer Available

tstana, 2023-10-05:
May be of general interest to update the list in the main README, a couple of devices no longer supported by manufacturers (EOL):

- [USH304](https://www.digikey.lu/en/products/detail/advantech-corporation/BB-USH304/7428806?s=N4IgTCBcDaIK4GcAWBmADAFhAXQL5A)
- [Delock 62537](https://www.delock.com/produkt/62537/merkmale.html?f=s)

mvp, 2023-10-07:
Perhaps half of the hub listed in compatibility table are not readily available, but often can be purchased on second hand market like eBay.

Do you have specific documentation or announcements from manufacturers about these products being discontinued?
E g. Delock hub was released very recently, in 2020. I highly doubt that is has been discontinued already. Unless proven otherwise, I'd keep it the way it is.

mvp, 2023-10-07:
Hmm, sorry, I jumped the gun too soon. Your links indeed prove that these products are EOL. I will update the table to mark 2023 as EOL for them. Thank you for reporting! @tstana

commit f37dcb5c (2023-10-07): Update EOL dates for some hubs

Updated EOL date for B+B SmartWorx USH304 to 2019.
https://www.mouser.com/PCN/Advantech_Corporation_EOL_BB_USH304.pdf.
Updated EOL date for Delock 62537, last available date seems to be 2021.

Closes #528.
closed 2023-10-07 by mvp