mvp/uhubctl #529: uhubctl works fine with 1 rpi4, but not with another four

trentbrown13, 2023-10-05:
Hello,
I have 5 x RPI 4 all running bookwork with the following bootloader and eeeprom
`i@BlueBerryPi:~ $ sudo rpi-eeprom-update 
BOOTLOADER: up to date
   CURRENT: Thu May 11 06:26:03 AM UTC 2023 (1683786363)
    LATEST: Thu May 11 06:26:03 AM UTC 2023 (1683786363)
   RELEASE: stable (/lib/firmware/raspberrypi/bootloader/stable)
            Use raspi-config to change the release.

  VL805_FW: Dedicated VL805 EEPROM
     VL805: up to date
   CURRENT: 000138c0
    LATEST: 000138c0
pi@BlueBerryPi:~ $ cat /proc/cpuinfo | grep Model
Model		: Raspberry Pi 4 Model B Rev 1.1
pi@BlueBerryPi:~ $ 
`
uhubctl works great with one of them, not on the other 4 resulting in this error 
`i@HalleBerryPi:~ $ sudo uhubctl -l 2 -a 0
Current status for hub 2 [1d6b:0003 Linux 6.1.21-v8+ xhci-hcd xHCI Host Controller 0000:01:00.0, USB 3.00, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 0203 power 5gbps U0 enable connect [0846:9060 MediaTek Inc. Wireless_Device 000000000]
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Sent power off request
New status for hub 2 [1d6b:0003 Linux 6.1.21-v8+ xhci-hcd xHCI Host Controller 0000:01:00.0, USB 3.00, 4 ports, ppps]
  Port 1: 0080 off
  Port 2: 0080 off
  Port 3: 0080 off
  Port 4: 0080 off
Current status for hub 1-1 [2109:3431 USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0503 power highspeed enable connect []
  Port 3: 0100 power
  Port 4: 0100 power

Message from syslogd@HalleBerryPi at Oct  5 14:32:07 ...
 kernel:[  909.372345] Internal error: Oops: 0000000096000005 [#1] PREEMPT SMP

Message from syslogd@HalleBerryPi at Oct  5 14:32:07 ...
 kernel:[  909.373117] Code: f90023f9 f9402278 941036ac f9402661 (39434020) 
^C
`

On the RPI that works it looks like this
`urrent status for hub 2 [1d6b:0003 Linux 6.1.21-v8+ xhci-hcd xHCI Host Controller 0000:01:00.0, USB 3.00, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Sent power off request
New status for hub 2 [1d6b:0003 Linux 6.1.21-v8+ xhci-hcd xHCI Host Controller 0000:01:00.0, USB 3.00, 4 ports, ppps]
  Port 1: 0080 off
  Port 2: 0080 off
  Port 3: 0080 off
  Port 4: 0080 off
Current status for hub 1-1 [2109:3431 USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0503 power highspeed enable connect [0846:9060 MediaTek Inc. Wireless_Device 000000000]
  Port 3: 0100 power
  Port 4: 0100 power
Sent power off request
New status for hub 1-1 [2109:3431 USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 1: 0000 off
  Port 2: 0000 off
  Port 3: 0000 off
  Port 4: 0000 off
Current status for hub 2 [1d6b:0003 Linux 6.1.21-v8+ xhci-hcd xHCI Host Controller 0000:01:00.0, USB 3.00, 4 ports, ppps]
  Port 1: 0080 off
  Port 2: 0080 off
  Port 3: 0080 off
  Port 4: 0080 off
Sent power on request
New status for hub 2 [1d6b:0003 Linux 6.1.21-v8+ xhci-hcd xHCI Host Controller 0000:01:00.0, USB 3.00, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 1-1 [2109:3431 USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 1: 0000 off
  Port 2: 0000 off
  Port 3: 0000 off
  Port 4: 0000 off
Sent power on request
New status for hub 1-1 [2109:3431 USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0101 power connect []
  Port 3: 0100 power
  Port 4: 0100 power
`

All 5 RPI's have the usb device plugged into the same USB port

Any ideas? Many thanks

trentbrown13, 2023-10-06:
Well, If I run it w/o the usb device inserted (Netgear nighthawk USB3 6ghz stick) it seems to work fine. Strange that it works with one nighthawk and not the other 4

i@BlackBerryPi:~ $ sudo uhubctl -l 2 -a 0
Current status for hub 2 [1d6b:0003 Linux 6.1.55-v8+ xhci-hcd xHCI Host Controller 0000:01:00.0, USB 3.00, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Sent power off request
New status for hub 2 [1d6b:0003 Linux 6.1.55-v8+ xhci-hcd xHCI Host Controller 0000:01:00.0, USB 3.00, 4 ports, ppps]
  Port 1: 0080 off
  Port 2: 0080 off
  Port 3: 0080 off
  Port 4: 0080 off
Current status for hub 1-1 [2109:3431 USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
Sent power off request
New status for hub 1-1 [2109:3431 USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 1: 0000 off
  Port 2: 0000 off
  Port 3: 0000 off
  Port 4: 0000 off

trentbrown13, 2023-10-06:
I may have resolved with the -p 100 flag, will test overnight

mvp, 2023-10-07:
You probably meant `-r 100` flag? Note that this flag should not be needed if you upgrade your OS to use Linux kernel 6.x.

Another question: which version of uhubctl you are using? It must be 2.4.0 or higher to work properly on RPi4B. Preferably 2/latest version 2.5.0.

trentbrown13, 2023-10-09:
Thank you mvp

no, I was using -p, I will try -r tomorrow. my linux kernel is 
6.1.21-v8+
and my uhubctl version is
pi@HalleBerryPi:~/bin/TEST $ uhubctl --version
2.5.0-1

I am still seeing issues but not as bad as before.
Thank you for the response!

mvp, 2023-10-12:
-r should not be needed under kernel 6.x, because uhubctl would use different method to turn power off, one that Linux kernel should specifically support since kernel 6.0. Perhaps some of your RPi4B instances were using kernel 5.x? That would explain small difference in their behavior.

Anyway, I am closing this since you seem to have a solution.

closed 2023-10-12 by mvp

trentbrown13, 2023-10-14:
Sorry for the late response. I checked and all my RPI4B's are using the 6.0 kernel and -r vs -p did not seem to make any difference. 

That said, I  don't think this is an issue with your code, likely more of an issue with the device as if I leave the device in the RPI4B but not doing anything it crashes on its own after some time.

I'll try another manufacture using the same Mediateck chip to see if there is a difference.
Thanks!