mvp/uhubctl #534: feature request: quiet output

3ricj, 2023-10-23:
feature request: 

--quiet -q suppress all messages except for critical errors

It's a lovely app which is working well, but boy it's chatty while doing it's work.  

Best, 
-3ric

3ricj, 2023-10-23:
frankly, the json output #124  would work for me as well.   It would mostly be nice to know if there was an error or not in the case of scripting.

mvp, 2023-10-25:
What exactly do you expect this new option to do? If you want to simply print nothing, you can already just redirect output to /dev/null, e.g.
```
uhubctl ... > /dev/null
```
If new -q option is enabled, I can't just completely avoid any printing to stdout, can I?

Regarding json output (not even talking that it's not trivial effort in it's own right), I would expect it will print even more text overall because output json most likely to have a lot more data in it than current output does.

3ricj, 2023-10-25:
As with many unix utilities, include classics like rsync and grep, there are options for verbosity and quiet.  Normally verbose (-v or -vv) provides additional details of what's going on -- helpful for debugging a problem.  

-q is common when "normal" output isn't needed, but you may want to see errors or exceptions.  I'm unsure if your app actually reports errors on stderr or stdout -- if errors are sent on stderr then redirection of stdout works great as a workaround. 

Effectively:  -q would be helpful if it only told me about problems.  If everything is great, then great, I don't need to see any output.  Much like I don't need a lightswitch that talks.   I mentioned json because I could filter out signals from noise much easier.  

Current, this is spewing into my logs.. which... is lovely but ...

Current status for hub 2 [1d6b:0003 Linux 6.1.21-v8+ xhci-hcd xHCI Host Controller 0000:01:00.0, USB 3.00, 4 ports, ppps]\n  Port 1: 0203 power 5gbps U0 enable connect [090c:1000 Samsung Flash Drive FIT 0307423070005202]\n  Port 2: 02a0 power 5gbps Rx.Detect\n  Port 3: 02a0 power 5gbps Rx.Detect\n  Port 4: 02a0 power 5gbps Rx.Detect\nSent power off request\nNew status for hub 2 [1d6b:0003 Linux 6.1.21-v8+ xhci-hcd xHCI Host Controller 0000:01:00.0, USB 3.00, 4 ports, ppps]\n  Port 1: 0080 off\n  Port 2: 0080 off\n  Port 3: 0080 off\n  Port 4: 0080 off\nCurrent status for hub 1-1 [2109:3431 USB2.0 Hub, USB 2.10, 4 ports, ppps]\n  Port 1: 0101 power connect []\n  Port 2: 0100 power\n  Port 3: 0100 power\n  Port 4: 0503 power highspeed enable connect [04a9:32f4 Canon Inc. Canon Digital Camera]\nSent power off request\nNew status for hub 1-1 [2109:3431 USB2.0 Hub, USB 2.10, 4 ports, ppps]\n  Port 1: 0000 off\n  Port 2: 0000 off\n  Port 3: 0000 off\n  Port 4: 0000 off\nCurrent status for hub 2 [1d6b:0003 Linux 6.1.21-v8+ xhci-hcd xHCI Host Controller 0000:01:00.0, USB 3.00, 4 ports, ppps]\n  Port 1: 0080 off\n  Port 2: 0080 off\n  Port 3: 0080 off\n  Port 4: 0080 off\nSent power on request\nNew status for hub 2 [1d6b:0003 Linux 6.1.21-v8+ xhci-hcd xHCI Host Controller 0000:01:00.0, USB 3.00, 4 ports, ppps]\n  Port 1: 02a0 power 5gbps Rx.Detect\n  Port 2: 02a0 power 5gbps Rx.Detect\n  Port 3: 02a0 power 5gbps Rx.Detect\n  Port 4: 02a0 power 5gbps Rx.Detect\nCurrent status for hub 1-1 [2109:3431 USB2.0 Hub, USB 2.10, 4 ports, ppps]\n  Port 1: 0000 off\n  Port 2: 0000 off\n  Port 3: 0000 off\n  Port 4: 0000 off\nSent power on request\nNew status for hub 1-1 [2109:3431 USB2.0 Hub, USB 2.10, 4 ports, ppps]\n  Port 1: 0100 power\n  Port 2: 0100 power\n  Port 3: 0100 power\n  Port 4: 0100 power