mvp/uhubctl #538: D-link DUB-H7/D1 (05E3:0608) works ! It is tested with USB tester, displaying volts and ampers !

CpServiceSpb, 2023-11-15:
I confirm that USB2 hub D-link DUB-H7/D1 based on 05E3:0608 plugged to USB3 host port work under Ubuntu 22.04 x64 fine per each port.
Tests were made by USB tester with oled displayng volts and ampers.
Test were made with 3g dongles, for example Huawei E1550 and Sentinel hardkeys.

When port is on, USB tester displays 3g dongle or hardkey power consumption.
And lsusb shows this device at list.

When port off command is sent display of USB tester is off as well as led of 3g dongle or hardkey plugged to the tester.
It mean that the power on this port is cut, that is port is off.
And lsusb doesn'  show this device at list.

There are 2 virtual hubs - 2 devpaths, in my case:
1. 2.4 - 2 digits;
2. 2.4.4 - 3 digits.

It means that 4th port of 1st hub is used for further 4 ports of other hub.
I did in the such way:
uhubctl -l 2.4 -p .... -a ...  - port from 1 to 3
uhubctl -l 2.4.4 -p .... -a ...  - port from 1 to 4

Pay in attention !
If you use 
uhubctl -l 2.4 -p 4 -a 0
all other 4 ports will be off
or if you use
uhubctl -l 2.4 -p 4 -a 1
all other 4 ports will be off.

That is 4th port in 2 digit hub is aplied for all 4 ports of 3 digit hub.

mvp, 2023-11-16:
Thanks for your report! This hub (DLink DUB-H7 rev D) is already listed as supported in device table, so no real action is necessary, but it is good to have another confirmation.

closed 2023-11-16 by mvp

CpServiceSpb, 2023-11-17:
I wrote because I did a test with a USB tester to look at if VBUS is really
cut or only up to 0.01A.

Regards ...

mjk-gh, 2024-01-08:
> Thanks for your report! This hub (DLink DUB-H7 rev D) is already listed as supported in device table, so no real action is necessary, but it is good to have another confirmation.

I am looking to buy such a USB hub, but I don't know how to find out the revision before buying.
Is the revision "D" indicated by the following strings on the label on the bottom?

P/N: ABCDE....**D**1E
[...]
H/W: **D**1

Or can you only find out the revision via software (lsusb?)?

Thanks in advance!

mvp, 2024-01-08:
@mjk-gh: unfortunately, it is really hard to tell if this particular hub will or will not work. I recommend buying it and if it doesn't work return it. I know it is a hassle, but we have seen manufacturers (especially DLink) to replace hubs internals without changing model numbers.

CpServiceSpb, 2024-01-08:
The revision is put at label on the. case.

Also I tested D2 in old case.
IIt works.

Regards, CpServiceSPb

mjk-gh, 2024-01-13:
> The revision is put at label on the. case.
Ah, good to know! Our hub seems to be labelled "A", and is
a 7-port hub in silver (which machtes the description in the
hardware compatibility table).

CpServiceSpb, 2024-01-14:
As I know power on ports of A revision is not cut for 100 %.
Look at the bottom of [](http://atc-ip.ru/info/29/1.html) on Russian language (you can use tranlator)
It was descrived revision A, as I understand.