mvp/uhubctl #539: crash under docker container

birosndr, 2023-11-16:
Hi,

I have a crash problem under docker container.
I first start the `docker run -ti -v /sys:/sys --privileged bash` container and under that run the uhubctl and I get the following error:
"Current status for hub 1 [1d6b:0002 Linux 5.10.107-android13-4-gd0319d1066a9-dirty dummy_hcd Dummy host controller dummy_hcd.0, USB 2.00, 1 ports, ppps]
  Port 1: 0100 power
Current status for hub 3 [1d6b:0003 Linux 5.10.107-android13-4-gd0319d1066a9-dirty xhci-hcd xHCI Host Controller xhci-hcd-exynos.4.auto, USB 3.10, 1 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
Current status for hub 2 [1d6b:0002 Linux 5.10.107-android13-4-gd0319d1066a9-dirty xhci-hcd xHCI Host Controller xhci-hcd-exynos.4.auto, USB 2.00, 1 ports, ppps]
  Port 1: 0503 power highspeed enable connect [0424:4216]
Segmentation fault (core dumped)"

First I want to know where can I found this core dump?
What other folder should I mount into the docker container?

Thank you,
Best regards.

birosndr, 2023-11-16:
I solved it!
you have to the docker run command the `-v /dev:/dev` because the uhubctl want to open files from there.

closed 2023-11-16 by birosndr

DeqingSun, 2023-11-28:
It seems mapping  /dev:/dev will work when the system is normal. 

However when some part of USB system crashes (still troubleshooting), uhubctl will still cause a crash. I don't remember the exact error code but it was caused by libusb lib and it may not be able to see the port. On host machine it was OK.

DeqingSun, 2023-12-01:
Here I'll summarize what I found and solved in the past 3 days. Hope this will help anyone in the future who want to use docker with uhubctl, or do hotplug/replug of USB devices in general in docker.

In general, docker will work well with any hardware already connected in host machine if the container in run in privileged mode and map /dev from host to container. However, once connected, the host and container will have /dev disconnected. And new devices in host will not be synced with container anymore so those devices will not be accessible to containers.

The host uses ```udev``` to deal with new devices, and the container can also use  ```udevadm monitor``` to observe all events, but I did not successfully got udev working in container or know anyone did it. So there need some work to do the udev job by ourselves.

There is a [balena](https://www.balena.io/) project solve the problem with a [script](https://github.com/balena-io-library/base-images/blob/master/examples/INITSYSTEM/systemd/systemd/entry.sh) to link ```/dev/ptmx``` in container. But doing so will [prevent the host machine from opening new terminals](https://forums.balena.io/t/udev-1-breaks-host-ubuntu-terminals/40619/3). Unless we run ```chmod 666 /dev/pts/ptmx``` in host after start up container. There is a [github issue](https://github.com/balena-io-library/base-images/issues/597) but no solution yet. I tested this approach and it worked. But I do not understand how it works so far, and I prefer not to do anything on the host that is easy to forget. 

I also see another [stack-overflow post](https://stackoverflow.com/questions/54802684/docker-not-seeing-usb-dev-ttyacm0-after-unplugging-and-then-replugging) suggesting doing ```mknod``` and ```rm``` for the ```/dev/ttyACM*``` devices in plug and unplug events. It worked well for my Arduinos but not for USB hubs. So if the hub got disconnected and reconnected for any reason. ```uhubctl``` will not work anymore. I found a [presentation slide](https://pic.huodongjia.com/ganhuodocs/2017-08-18/1503028636.63.pdf) and on page 57, it mentioned a way to use docker to access hot-plugged smartphones for testing. It seems the method also work for USB hub. 

So in my application, I have a python script constantly listening to output of  ```udevadm monitor```. If there is new ```add``` event of hub or serial device, it will run ```mknod``` to create the node. And those node got removed in ```remove``` event.

Here I attached a minimal docker environment that works with hot-plugging CH335F HUB and Arduino Zero. There is more detailed comment in it.

[usbPlugTest.zip](https://github.com/mvp/uhubctl/files/13520225/usbPlugTest.zip)