mvp/uhubctl #540: Cannot cut power on USB3.0

Enohriel, 2023-11-16:
Hello,

I'm using a PCIe USB card :
```
Current status for hub 2 [1d6b:0003 Linux 5.10.0-26-amd64 xhci-hcd xHCI Host Controller 0000:04:00.0, USB 3.00, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 1 [1d6b:0002 Linux 5.10.0-26-amd64 xhci-hcd xHCI Host Controller 0000:04:00.0, USB 2.00, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0503 power highspeed enable connect [046d:0825 05CFC1B0]
  Port 4: 0000 off
```
I can easily cut off power for USB2 vhub :
```
# uhubctl -a off -l 1 -p 4
Current status for hub 1 [1d6b:0002 Linux 5.10.0-26-amd64 xhci-hcd xHCI Host Controller 0000:04:00.0, USB 2.00, 4 ports, ppps]
  Port 4: 0000 off
Sent power off request
New status for hub 1 [1d6b:0002 Linux 5.10.0-26-amd64 xhci-hcd xHCI Host Controller 0000:04:00.0, USB 2.00, 4 ports, ppps]
  Port 4: 0000 off
```
But when I try to cut power from USB3.0 vhub, it says that the device is not compatible.

```
root@Plex2:~# uhubctl -a off -l 2 -p 4
No compatible devices detected at location 2!
```

Isn't it weird ? Because it shows under listing.

mvp, 2023-11-16:
Make sure you use uhubctl version 2.5.0 or later. Versions older than 2.4.0 may not work correctly with USB 3.

Enohriel, 2023-11-16:
I was using Debian package, which is version 2.3.0.

I just compiled the latest but result is the same :

```
# uhubctl --version
2.5.0-24-gf37dcb5c

# uhubctl
Current status for hub 3-4 [0e0f:0002 VMware, Inc. VMware Virtual USB Hub, USB 2.00, 7 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
  Port 5: 0100 power
  Port 6: 0100 power
  Port 7: 0100 power
Current status for hub 3-3 [0e0f:0002 VMware, Inc. VMware Virtual USB Hub, USB 1.10, 7 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
  Port 5: 0100 power
  Port 6: 0100 power
  Port 7: 0100 power
Current status for hub 2 [1d6b:0003 Linux 5.10.0-26-amd64 xhci-hcd xHCI Host Controller 0000:04:00.0, USB 3.00, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 1 [1d6b:0002 Linux 5.10.0-26-amd64 xhci-hcd xHCI Host Controller 0000:04:00.0, USB 2.00, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0503 power highspeed enable connect [046d:0825 05CFC1B0]
  Port 4: 0000 off

# uhubctl -a off -l 2 -p 4
No compatible devices detected at location 2!
```

mvp, 2023-11-16:
This is weird. You may need to run this under debugger, e.g.

```
make
gdb ./uhubctl
set args <command line params>
b <line number>
c
p <variable>
l  <list source>
```
I remember having similar issues few years ago when USB duality matching didn't work due to some obscure problem, like dual hub showing as ganged vs main being ppps.
Also, can you show output of `uhubctl -f`?
And perhaps try `uhubctl -a off -l 2 -p 4 -f`?

Enohriel, 2023-11-16:
With -f :

```
# uhubctl -f
Current status for hub 4 [1d6b:0003 Linux 5.10.0-26-amd64 xhci-hcd xHCI Host Controller 0000:1b:00.0, USB 3.10, 4 ports, nops]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 3-4 [0e0f:0002 VMware, Inc. VMware Virtual USB Hub, USB 2.00, 7 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
  Port 5: 0100 power
  Port 6: 0100 power
  Port 7: 0100 power
Current status for hub 3-3 [0e0f:0002 VMware, Inc. VMware Virtual USB Hub, USB 1.10, 7 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
  Port 5: 0100 power
  Port 6: 0100 power
  Port 7: 0100 power
Current status for hub 3 [1d6b:0002 Linux 5.10.0-26-amd64 xhci-hcd xHCI Host Controller 0000:1b:00.0, USB 2.00, 4 ports, nops]
  Port 1: 0103 power enable connect [0e0f:0003 VMware VMware Virtual USB Mouse]
  Port 2: 0100 power
  Port 3: 0107 power suspend enable connect [0e0f:0002 VMware, Inc. VMware Virtual USB Hub, USB 1.10, 7 ports, ppps]
  Port 4: 0507 power highspeed suspend enable connect [0e0f:0002 VMware, Inc. VMware Virtual USB Hub, USB 2.00, 7 ports, ppps]
Current status for hub 2 [1d6b:0003 Linux 5.10.0-26-amd64 xhci-hcd xHCI Host Controller 0000:04:00.0, USB 3.00, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 1 [1d6b:0002 Linux 5.10.0-26-amd64 xhci-hcd xHCI Host Controller 0000:04:00.0, USB 2.00, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0503 power highspeed enable connect [046d:0825 05CFC1B0]
  Port 4: 0000 off
# uhubctl -a off -l 2 -p 4 -f
No compatible devices detected at location 2!
Run with -h to get usage info.
```

Absolutly no idea how to run it under debugger.

Here is my poor try :
```
# gdb ./uhubctl
GNU gdb (Debian 10.1-1.7) 10.1.90.20210103-git
Copyright (C) 2021 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./uhubctl...
(gdb) set args -a off -l 2 -p 4
(gdb) b
No default breakpoint address now.
(gdb) b 1
Breakpoint 1 at 0x2472: file uhubctl.c, line 62.
(gdb) c
The program is not being run.
(gdb) p
The history is empty.
(gdb) l
1017                    "There were permission problems while accessing USB.\n"
1018                    "Follow https://git.io/JIB2Z for a fix!\n"
1019                );
1020            }
1021    #endif
1022            return LIBUSB_ERROR_ACCESS;
1023        }
1024        return hub_phys_count;
1025    }
1026
```

mvp, 2023-11-16:
It's very hard to tell without having local access, but I think duality matching is getting confused. It seems like it thinks that USB3 hub `2` doesn't have dual USB2 hub to it.
As a workaround, you can use -e per each dual hub e.g.
```
uhubctl -a off -l 2 -p 4 -e
uhubctl -a off -l 1 -p 4 -e
```

Speaking of debugging, you did pretty well for first attempt! You can set breakpoints per function name or line number as printed by l.
Perhaps try setting breakpoint to main (`b main`) and see what code is trying to do. `c` (continue) will keep running until it encounters active breakpoint

Enohriel, 2023-11-16:
I've already tried with -e, and got the same results.
I just tried again and got a different result tho, maybe because I just messed with udev, or maybe I misspelled the first time.

```root@Plex2:~/uhubctl# uhubctl -a off -l 2 -p 4 -e
Current status for hub 2 [1d6b:0003 Linux 5.10.0-26-amd64 xhci-hcd xHCI Host Controller 0000:04:00.0, USB 3.00, 4 ports, ppps]
  Port 4: 02a0 power 5gbps Rx.Detect
Sent power off request
New status for hub 2 [1d6b:0003 Linux 5.10.0-26-amd64 xhci-hcd xHCI Host Controller 0000:04:00.0, USB 3.00, 4 ports, ppps]
  Port 4: 0080 off
root@Plex2:~/uhubctl# uhubctl
Current status for hub 3-4 [0e0f:0002 VMware, Inc. VMware Virtual USB Hub, USB 2.00, 7 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
  Port 5: 0100 power
  Port 6: 0100 power
  Port 7: 0100 power
Current status for hub 3-3 [0e0f:0002 VMware, Inc. VMware Virtual USB Hub, USB 1.10, 7 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
  Port 5: 0100 power
  Port 6: 0100 power
  Port 7: 0100 power
Current status for hub 2 [1d6b:0003 Linux 5.10.0-26-amd64 xhci-hcd xHCI Host Controller 0000:04:00.0, USB 3.00, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 0080 off
Current status for hub 1 [1d6b:0002 Linux 5.10.0-26-amd64 xhci-hcd xHCI Host Controller 0000:04:00.0, USB 2.00, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0503 power highspeed enable connect [046d:0825 05CFC1B0]
  Port 4: 0000 off
```

Unfortunately, USB port still have power !

```
(gdb) set args -a off -l 2 -p 4
(gdb) b main
Breakpoint 1 at 0x42c0: file uhubctl.c, line 1031.
(gdb) c
The program is not being run.
(gdb) run
Starting program: /root/uhubctl/uhubctl -a off -l 2 -p 4
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".

Breakpoint 1, main (argc=7, argv=0x7fffffffeb68) at uhubctl.c:1031
1031        int c = 0;
(gdb) c
Continuing.
[New Thread 0x7ffff7d7c700 (LWP 150542)]
No compatible devices detected at location 2!
Run with -h to get usage info.
[Thread 0x7ffff7d7c700 (LWP 150542) exited]
[Inferior 1 (process 150536) exited with code 01]
(gdb) p
The history is empty.
(gdb) l
1026
1027
1028    int main(int argc, char *argv[])
1029    {
1030        int rc;
1031        int c = 0;
1032        int option_index = 0;
1033
1034        for (;;) {
1035            c = getopt_long(argc, argv, short_options, long_options, &option_index);
(gdb)
```

mvp, 2023-11-17:
If ports still have power, it simply means your hardware is not compliant. Or, reports supporting per port power switching, but in fact only doing ganged - who knows.

Closing this as not actionable from uhubctl standpoint

closed 2023-11-17 by mvp