mvp/uhubctl #543: Radxa Rock Pi S Compatibility?

NadavK, 2023-11-19:
Can you kindly tell if your wonderful utility is compatible with Radxa Rock Pi S?

```
$ lsusb
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 002 Device 004: ID 1286:4e31 Marvell Semiconductor, Inc. Mobile Composite Device Bus
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

// trying to power off Bus 002 Device 004

$ sudo uhubctl -n 1d6b -p 1 -a 0 -l 2
Current status for hub 2 [1d6b:0002 Linux 6.1.32-rockchip64 ehci_hcd EHCI Host Controller ff440000.usb, USB 2.00, 1 ports, ppps]
  Port 1: 0503 power highspeed enable connect [1286:4e31 Marvell Mobile Composite Device Bus 200806006809080000]
Sent power off request
New status for hub 2 [1d6b:0002 Linux 6.1.32-rockchip64 ehci_hcd EHCI Host Controller ff440000.usb, USB 2.00, 1 ports, ppps]
  Port 1: 0000 off

```
The output indicate that it is powering off, but in reality nothing happens.

mvp, 2023-11-19:
1. Make sure you are using latest uhubctl version (2.5.0 as of this writing).
2. Not all devices are advertising per port power switching. And even when they do (like yours), it doesn't necessarily mean that they actually support it.
3. If hardware doesn't support turning power off, there is not really much that software can do. Rarely, there were bugs with uhubctl when it was software issue (e.g. when I was adding USB3 support back in 2020), but if you have strictly USB2 hardware it was working correctly from the very first version.

closed 2023-11-19 by mvp

NadavK, 2023-11-21:
@mvp, thank you for your quick reply.
I meant to ask, if I am using the correct parameter values when calling uhubctl?

mvp, 2023-11-21:
Your uhubctl parameters look fine. You don't really need to specify -n as long as you provide location (-l), port (-p) and action (-a)