mvp/uhubctl #547: Raspberry Pi5 Raspbian 12 bookworm

wlanrouter, 2023-12-01:
It seems like uhubctl ist not working on Raspberry Pi5 Raspbian 12 bookworm

```
# lsusb
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 012: ID 0403:6001 Future Technology Devices International, Ltd FT232 Serial (UART) IC
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

```
# uhubctl
Current status for hub 4 [1d6b:0003 Linux 6.1.0-rpi6-rpi-v8 xhci-hcd xHCI Host Controller xhci-hcd.1, USB 3.00, 1 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
Current status for hub 3 [1d6b:0002 Linux 6.1.0-rpi6-rpi-v8 xhci-hcd xHCI Host Controller xhci-hcd.1, USB 2.00, 2 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
Current status for hub 2 [1d6b:0003 Linux 6.1.0-rpi6-rpi-v8 xhci-hcd xHCI Host Controller xhci-hcd.0, USB 3.00, 1 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
Current status for hub 1 [1d6b:0002 Linux 6.1.0-rpi6-rpi-v8 xhci-hcd xHCI Host Controller xhci-hcd.0, USB 2.00, 2 ports, ppps]
  Port 1: 0100 power
  Port 2: 0103 power enable connect [0403:6001 FTDI FT232R USB UART 001X8MQT]
```

I want to disable FT232 as I am used to from my Raspberry Pi 3B

```
# uhubctl -a 0 -l 1 -p 2
Current status for hub 1 [1d6b:0002 Linux 6.1.0-rpi6-rpi-v8 xhci-hcd xHCI Host Controller xhci-hcd.0, USB 2.00, 2 ports, ppps]
  Port 2: 0103 power enable connect [0403:6001 FTDI FT232R USB UART 001X8MQT]
Sent power off request
New status for hub 1 [1d6b:0002 Linux 6.1.0-rpi6-rpi-v8 xhci-hcd xHCI Host Controller xhci-hcd.0, USB 2.00, 2 ports, ppps]
  Port 2: 0000 off
```
```
# uhubctl
Current status for hub 4 [1d6b:0003 Linux 6.1.0-rpi6-rpi-v8 xhci-hcd xHCI Host Controller xhci-hcd.1, USB 3.00, 1 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
Current status for hub 3 [1d6b:0002 Linux 6.1.0-rpi6-rpi-v8 xhci-hcd xHCI Host Controller xhci-hcd.1, USB 2.00, 2 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
Current status for hub 2 [1d6b:0003 Linux 6.1.0-rpi6-rpi-v8 xhci-hcd xHCI Host Controller xhci-hcd.0, USB 3.00, 1 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
Current status for hub 1 [1d6b:0002 Linux 6.1.0-rpi6-rpi-v8 xhci-hcd xHCI Host Controller xhci-hcd.0, USB 2.00, 2 ports, ppps]
  Port 1: 0100 power
  Port 2: 0000 off
```

```
# lsusb
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 013: ID 0403:6001 Future Technology Devices International, Ltd FT232 Serial (UART) IC
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

The device LED is also powered. Is it Raspberry that removed the feature from Pi5? Can believe it, but I am also not able to find information regarding the hardware support.

nbuchwitz, 2024-01-18:
I did some investigation with RPi5 and uhubctl. Looks like the port are kinda ganged / the driver behaves oddly: If you disable USB3 duality handling (`-e`) and turn off each port individually the last port switches the power off (order does not matter). The first port which is activated turns the power back on (also order does not matter). 

Will dig a bit deeper in the RP1 driver and let you know if I found something.

nbuchwitz, 2024-01-20:
Long story short: Raspberry Pi 5 does only have one vbus switch (ganged) and thus cannot support per port power switching. If you switch off all ports, the last clear feature command switches off the power. The first set feature command turns it on again (like specified in the usb specs). 

See https://github.com/raspberrypi/linux/issues/5870#issuecomment-1900684562

mvp, 2024-01-21:
Thanks for investigating @nbuchwitz. Can you please show output for `sudo uhubctl` on RPi5, what it's USB topology looks like, which ports are required to be turned off, etc?

Basically, I want to create a doc page similar to what we have for RPi4B: https://github.com/mvp/uhubctl#raspberry-pi-4b.

nbuchwitz, 2024-01-21:
Can post this later / already been working on a doc or for this

mvp, 2024-04-10:
Again, unfortunately I don't have Raspberry Pi 5 in my possession and cannot confirm it one way or another.
I suspect we may need some hack implemented in uhubctl to make RPi5 work,
much like what we did for RPi4.

nbuchwitz, 2024-04-10:
Try to find some spare time for my pr on this. As I already wrote above, the implementation on Pi5 is unfortunately a bit broken. There is only one ganged vbus switch, which disables all ports at once. This means that individual port vusb control cannot work on Pi5.

wlanrouter, 2024-04-10:
How can I disable all ports? There is only one device attached, so I could live with disabling all ports.

rmorgans, 2024-04-15:
@mvp reach out if you want a pi5, email address in my github profile.

mvp, 2024-04-15:
Thank you @rmorgans! Responded

mvp, 2024-04-19:
Added support for Raspberry Pi 5 in 3d804c1ab7479.

Much thanks to @rmorgans for sending me new Raspberry Pi 5 so I was able to debug the issue.

Read more about RPi5 USB topology here https://github.com/mvp/uhubctl?tab=readme-ov-file#raspberry-pi-5,
but long story short, build uhubctl from master branch, then use these 2 commands to turn power off (all ports only):
```
uhubctl -l 1 -a 0
uhubctl -l 3 -a 0
```
and this to turn back on:
```
uhubctl -l 1 -a 1
uhubctl -l 3 -a 1
```

commit 3d804c1a (2024-04-19): Add support for Raspberry Pi 5

Added hack to support Raspberry Pi 5.
Updated README with recipe how to control power on RPi5.

Closes #547.
closed 2024-04-19 by mvp