mvp/uhubctl #554: uhubctl confuses different hubs on the same box

fpoto, 2024-01-30:
I have a motherboard Motherboard Asus ROG Crosshair X670E Hero. This is what I see:
```
# uhubctl
Current status for hub 6-5 [174c:3074 ASUS TEK. ASM107x, USB 3.00, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 5-5 [174c:2074 ASUS TEK. ASM107x, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0503 power highspeed enable connect [058f:6387 Generic Mass Storage 83F6ED99]
  Port 3: 0100 power
  Port 4: 0100 power
```
So there are two hubs, called 5-5 (usb2) and 6-5 (usb3). Just for testing, I have put a usb memory key on port 2 of the usb2 hub.

Now, if I try to select only port 2 of hub 5-5, I see that uhubctl confuses the two hubs:
```
# uhubctl -s Mass
Current status for hub 6-5 [174c:3074 ASUS TEK. ASM107x, USB 3.00, 4 ports, ppps]
  Port 2: 02a0 power 5gbps Rx.Detect
Current status for hub 5-5 [174c:2074 ASUS TEK. ASM107x, USB 2.10, 4 ports, ppps]
  Port 2: 0503 power highspeed enable connect [058f:6387 Generic Mass Storage 83F6ED99]
```
that is, instead of showing only port 2 from 5-5, it shows port 2 of both hubs. More:
```
# uhubctl -l 5-5 -p 2
Current status for hub 6-5 [174c:3074 ASUS TEK. ASM107x, USB 3.00, 4 ports, ppps]
  Port 2: 02a0 power 5gbps Rx.Detect
Current status for hub 5-5 [174c:2074 ASUS TEK. ASM107x, USB 2.10, 4 ports, ppps]
  Port 2: 0503 power highspeed enable connect [058f:6387 Generic Mass Storage 83F6ED99]

# uhubctl -l 5-5.2
No compatible devices detected at location 5-5.2!
Run with -h to get usage info.
```
This is not for port 2 only:
```
# uhubctl -l 6-5
Current status for hub 6-5 [174c:3074 ASUS TEK. ASM107x, USB 3.00, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 5-5 [174c:2074 ASUS TEK. ASM107x, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0503 power highspeed enable connect [058f:6387 Generic Mass Storage 83F6ED99]
  Port 3: 0100 power
  Port 4: 0100 power

# uhubctl -l 5-5
Current status for hub 6-5 [174c:3074 ASUS TEK. ASM107x, USB 3.00, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 5-5 [174c:2074 ASUS TEK. ASM107x, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0503 power highspeed enable connect [058f:6387 Generic Mass Storage 83F6ED99]
  Port 3: 0100 power
  Port 4: 0100 power
```

mvp, 2024-01-30:
This works as designed. Please read this to understand why it works this way: https://github.com/mvp/uhubctl#usb-30-duality-note

Basically, uhubctl by default would show not only hub you are filtering, but also hub that is dual to it (per USB3/USB2 duality). This is because in order to truly turn power off, it must turn power off for both dual hubs.

If you insist, you can use -e to disable duality handling, but by doing so you will not actually turn power off.

closed 2024-01-30 by mvp

fpoto, 2024-01-31:
Thank you for pointing me to the docs. I probably had read it but at the time could not understand its meaning.

I see that my hub is not listed among the working ones. I will check that everything works and will report on it.