mvp/uhubctl #559: USB3-HUB7-81X not working

ben-burton, 2024-03-23:
I managed to track down and purchased a USB3-HUB7-81X-EU which may be different variation to the USB3-HUB7-81X listed. On the device itself the model number is the same USB3-HUB7-81X but has a serial number with the additional -EU qualifier (I am based in the UK).
_uhubctl is reporting 2109:0812 instead of 2109:0813._ 

It is recognised by uhubctl and I was able to turn off data connection to all ports and blue lights when off on each port, but power still remained to the usb fan I tested with. Shame I couldn't get this one working :(

Thought I would share my experience in case it's useful info.

mvp, 2024-05-21:
Per comment in README:
```
USB3-HUB7-81X (only works for 2 charge ports)
```
@ben-burton can you please test if it works for charge ports?

ben-burton, 2024-05-21:
I couldn’t get it to work with any of the ports on the usb hub. Please help me to understand if I’ve misunderstood what you meant

mvp, 2024-05-21:
Thanks for your report @ben-burton, I have removed this hub from supported list.
I want to keep only reliably working hubs in the list.

commit 83f624f3 (2024-05-21): Removed Plugable USB3-HUB7-81X from supported list

Closes #559.