mvp/uhubctl #565: CoolGear Hub control (CG-10PU3MGD) 5V rail not switching

mooneyj, 2024-05-05:
The CG-10PU3MGD hub can be controlled with uhubctl, with devices being disconnected from host. However the 5V rail does not switch off.

There is a binary blob available from the hub manufacturers which can switch the 5V rail per port. I have tested this.
This binary is called cusbi and is available to download from the [manufacturers website](https://www.coolgear.com/product/10-port-managed-usb-3-hub-w-15kv-esd-surge-protection-on-off-per-port-control-software).

Perhaps we can reverse engineer the binary to allow uhubctl to control coolgear hubs?

If anyone is up for helping me with this I can provide some hardware testing. I might also reach out to coolgear to guage their appetite for helping.

mvp, 2024-05-10:
Apparently, Coolgear decided to use USB serial connection as their management interface, and not bother making their hu lb to properly support USB per port power switching as defined by USB 2.0 standard.

I understand why they made use of USB serial - it works on any operating system, including Windows - uhubctl doesn't work on Windows due to winusb.sys not allowing direct access to USB hubs. However, that should not be the reason to not support PPPS for all other OS in addition to serial control.

If Coolgear had documentation about their serial control protocol, I don't think it is very difficult to add support for it. However, such addition is better suited to be added to [uhidctl](https://github.com/mvp/uhidctl) - another utility I wrote to control power for cheap power relays that have HID USB interface (almost the same as serial).

mvp, 2024-05-21:
Since there does not seem to be viable solution for this or interest from Coolgear to add standard's compliant power switching, I will close this.
Feel free to reopen if you get any news from Coolgear.

closed 2024-05-21 by mvp

mooneyj, 2024-05-21:
They've confirmed that they won't be able to modify their hubs or uhubctl software to support it (some of their hubs do support uhubctl). Their cusbi binary has been working fine for my application. 

Thanks for looking into this folks.

For the record here's their response:

> We are aware of the uhubctl software, and although we do not officially support it, I have previously tested a select few products with the software.
> 
> Hubs tested:
> 
> CG43AB1VV - Does support PPPS, can be controlled using uhubctl
> 
> CG-7U3DS - Does support PPPS, can be controlled using uhubctl
> 
> USBG- 7U2ML - Not "supported" by uhubctl but using -f does result in data connection being cutoff, but the port still receives power. Not officially PPPS capable.
> 
> CG-U3MINI4P-BPH - Not "supported" by uhubctl. Using -f results in both power and data being cutoff.
> 
> CG-U3MICRO4PH - Does support PPPS, can be controlled with uhubctl, ports properly turn off power and data with normal uhubctl commands.
> 
> Overall, uhubctl is compatible with some of our products.
> However, we do not officially support this software and cannot offer technical support when it is used in conjunction with our hubs due to the lack of extensive, overall testing with the hubs.
> Unfortunately due to NDA agreements and development time and costs, we would not be able to revise specific hubs to make them more compatible specifically for uhubctl, nor will we be able to add to the uhubctl codebase to make itself more compatible with our hubs.