mvp/uhubctl #571: VirtualHere PPPS USB Hub

virtualhere, 2024-05-30:
Hi,  I created a PPPS USB 4-port hub that can be purchased here https://modularkvmip.com/product/usb-hub/

Can you add it to the list of supported PPPS hubs.

I designed it myself to fully support per -port power switching and i made sure it has a unique ContainerID ! So there is no confusion finding the dual hub. I also added an aluminium shell so its pretty strong.

You can  turn it on and off with my virtualhere software or just standard uhubctl commands. 

I also wrote a windows command line program to turn it on and off on windows hosts using a windows filter i wrote.

I sell a whole KVM/IP kit i just created - but just the hub can be ordered and shipped.

Michael

virtualhere, 2024-06-05:
I wanted to add, I'm happy to open-source the hub filter code if you are interested in adding it to this repository so that you can control power on Windows for any PPPS supported hub. I can sign the code that would be posted on this website with my company EV certificate so it will work with all Windows versions > 10

closed 2024-06-05 by mvp

mvp, 2024-06-05:
Sorry, this slipped through cracks - I have added your hub to supported list now!

Regarding Windows filter: uhubctl is designed to support as many devices as possible, the only requirement being supporting PPPS. I love idea of being able to support Windows, but it would not be generic if you used some custom protocol (e.g. HID communication to your hub).

That said, if it is possible to make your hub filter code generic and be able to support most existing PPPS hubs under Windows, I would be excited to add official Windows support into uhubctl. Of course it would require installing your signed filter driver, but that is a lot better that not supporting it on Windows at all.

We had an effort to develop generic Windows filter 6 years ago in https://github.com/mvp/uhubctl/issues/79, but that work got shelved.

Anyway, let me know what protocol you are using and I would be happy to add Windows support to uhubctl if it is generic enough. Thank you @virtualhere!

mvp, 2024-06-05:
Few more comments: awesome work on paying attention to details like unique ContainerID 👍 

I have small concern on power. It looks like your hub is bus-powered? I don't see it having external power supply? This could potentially be an issue, since upstream USB will provide 500mA (900mA in USB3 mode), while each downstream port may consume up to 900mA. This probably would work fine if downstream ports use lightweight clients like keyboard/mouse/security dongle. But it would most likely not work well if there are even 2 heavy downstream devices (e.g. 2 USB hard drives).

In my experience, all well supported hubs do have external power supply. It is often optional, but is usually required to make PPPS work really well.

virtualhere, 2024-06-05:
1. Regarding the filter, i wrote it so its generic, it works with any hub. It uses the normal USB control request to set PORT_POWER, it does not use any custom HID commands. I will need to modify the filter on my website slightly for it to be GPL (it does a bit more than needed at the moment)

2. I found instead of a separate power supply its easier to just use a USB "y-cable" and connect that second end to another usb port or power supply if its a problem. The hub will advertise as "self-powered" and so will pull more current than 900ma if the host controller allows it (most newer ones do). This way its more flexible.

mvp, 2024-06-06:
1. That is really awesome news on your filter being generic! I would be happy to make your code cleaner and publish it as part of uhubctl repository (of course attributing you). Or, feel free to raise PR to uhubctl with your changes.

2. Thanks for the Y-cable with extra power source tip! I did not know that this could possibly work. I guess this depends on USB hub chip to be quite permissive.

virtualhere, 2024-06-06:
OK i will send you the driver code today after i simplify and secure it.

virtualhere, 2024-06-06:
Here is the userspace part [pppshubfilterexe.zip](https://github.com/user-attachments/files/15613446/pppshubfilterexe.zip)
Here is the driver part [pppshubfilter.zip](https://github.com/user-attachments/files/15613469/pppshubfilter.zip)
Here is the compiled exe and sys signed  [test.zip](https://github.com/user-attachments/files/15613492/test.zip)

It might be hard to add to uhubctl because it uses nothing from libusb etc as its not relevant.
If you want to test expand the exe/sys zip and then run
pppshubfilter.exe --install-filter

The install-filter argument will install the filter driver but will try not to require a reboot. However i recommend a reboot anyway after the filter is installed because sometimes windows acts a bit weird when you install a filter dynamically and dont reboot..it acts like the filter is not there for the very first call

mvp, 2024-06-07:
Thank you for sending source code. I would like to add user space windows support in uhubctl itself. Your user space client makes use of Windows specific constructs. Instead, I would rather have the same device addressing that uhubctl is using on all OS.

Basically, uhubctl on Windows should just work as usual, supporting (almost) all command line options and overall operating the same, with the only requirement is having your kernel filter installed.

Granted, you made installing kernel filter easier by making user space client to automate this process. This is nice and perhaps can be added to uhubctl as well, but I don't think this is critical - we can ask users to install filter separately.

At any rate, @virtualhere you made significant progress on Windows support - much thanks for that! I will open another issue to track status of Windows support.

I will try to find some time to work on this, but it won't be very quick - may take weeks.

virtualhere, 2024-06-07:
Once you do an initial check-in of the windows code, i will help modify it to work with uhubctl args

msbroadf, 2024-10-19:
On the readme.md could you add a note saying virtualhere hub also supports windows.

virtualhere, 2024-10-19:
Could you add a note on the README.md saying that the virtualhere hub also supports windows

diegocr, 2024-12-23:
@virtualhere can you sell your USB Hub on Ebay, or otherwise accept PayPal payments in your site, please? 🙂

ABelliqueux, 2025-07-09:
So what is the state of compatibility with this HUB as of today ? Is it fully compatible with uhubctl on Linux ?

ABelliqueux, 2025-07-10:
🥳  Nice, thanks for your replies !