mvp/uhubctl #572: Working Hub: ikuai Aluminium 7-Port USB Hub

samlii, 2024-06-19:
Thanks for this project, was exactly what I was looking for.

I recently bought the ikuai Aluminium 7-Port Powered USB 3.0 Hub from [Amazon](https://www.amazon.com/gp/product/B0BJJGYK2W).  It works perfectly for what I am using it for (controlling a light bar over a 3d printer).  The ports have LEDs to indicate if they are on and those also show when the status is toggles.  

It does show up as separate USB 2 and 3 hubs, but like I said seems to work fine. 
One thing to note, the ports by default come with the power off, took me a while to realize this :)

vID:pID
---------------------
0bda:5411 Realtek Semiconductor Corp. RTS5411 Hub - USB 2.1
0bda:0411 Realtek Semiconductor Corp. Hub - USB 3.2

dmesg
----------------------
[    1.981013] usb 4-2: new SuperSpeed USB device number 2 using xhci_hcd
[    2.016551] usb 4-2: New USB device found, idVendor=0bda, idProduct=0411, bcdDevice= 0.04
[    2.016554] usb 4-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    2.016556] usb 4-2: Product: USB3.2 Hub
[    2.016557] usb 4-2: Manufacturer: Generic
[    2.021492] hub 4-2:1.0: USB hub found
[    2.024082] hub 4-2:1.0: 4 ports detected
[    2.136317] usb 3-2: new high-speed USB device number 3 using xhci_hcd
[    2.297463] usb 3-2: New USB device found, idVendor=0bda, idProduct=5411, bcdDevice= 0.04
[    2.297466] usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    2.297468] usb 3-2: Product: USB2.1 Hub
[    2.297469] usb 3-2: Manufacturer: Generic
[    2.299212] hub 3-2:1.0: USB hub found
[    2.300380] hub 3-2:1.0: 4 ports detected
--
[    2.372991] usb 4-2.4: new SuperSpeed USB device number 3 using xhci_hcd
[    2.408535] usb 4-2.4: New USB device found, idVendor=0bda, idProduct=0411, bcdDevice= 0.04
[    2.408540] usb 4-2.4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    2.408542] usb 4-2.4: Product: USB3.2 Hub
[    2.408544] usb 4-2.4: Manufacturer: Generic
[    2.415991] hub 4-2.4:1.0: USB hub found
[    2.417990] hub 4-2.4:1.0: 4 ports detected
--
[    3.884341] usb 3-2.4: new high-speed USB device number 10 using xhci_hcd
[    3.996837] usb 3-2.4: New USB device found, idVendor=0bda, idProduct=5411, bcdDevice= 0.04
[    3.996843] usb 3-2.4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    3.996845] usb 3-2.4: Product: USB2.1 Hub
[    3.996846] usb 3-2.4: Manufacturer: Generic
[    3.998603] hub 3-2.4:1.0: USB hub found
[    3.999732] hub 3-2.4:1.0: 4 ports detected

lsusb output
-----------------
Bus 004 Device 003: ID 0bda:0411 Realtek Semiconductor Corp. Hub
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.20
  bDeviceClass            9 Hub
  bDeviceSubClass         0
  bDeviceProtocol         3
  bMaxPacketSize0         9
  idVendor           0x0bda Realtek Semiconductor Corp.
  idProduct          0x0411 Hub
  bcdDevice            0.04
  iManufacturer           1 Generic
  iProduct                2 USB3.2 Hub
  iSerial                 0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x001f
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
[... 395 lines cut ...]
    bmAttributes   0x0000f41e
      BESL Link Power Management (LPM) Supported
    BESL value     1024 us
    Deep BESL value    61440 us
  SuperSpeed USB Device Capability:
    bLength                10
    bDescriptorType        16
    bDevCapabilityType      3
    bmAttributes         0x00
    wSpeedsSupported   0x000e
      Device can operate at Full Speed (12Mbps)
      Device can operate at High Speed (480Mbps)
      Device can operate at SuperSpeed (5Gbps)
    bFunctionalitySupport   1
      Lowest fully-functional device speed is Full Speed (12Mbps)
    bU1DevExitLat          10 micro seconds
    bU2DevExitLat        1023 micro seconds
  Container ID Device Capability:
    bLength                20
    bDescriptorType        16
    bDevCapabilityType      4
    bReserved               0
    ContainerID             {20b9cde5-7039-e011-a935-0002a5d5c51b}
Device Status:     0x0001
  Self Powered

mvp, 2024-08-05:
This hub: [ikuai A107-5](https://www.amazon.com/dp/B0BJJGYK2W) is made by Shenzhen Rong Sheng Hui (aka RSHTech, aka Rosonway), appears to be identical to [Rosonway RSH-A107](https://www.amazon.com/dp/B0BTNTZ4BW).

Perhaps we should add it as one device into the table?

samlii, 2024-08-05:
I think that makes sense.

closed 2024-08-31 by mvp

WinkelCode, 2024-09-03:
The 4-port version supposedly only properly turns off devices connected only via USB 2.0 (https://github.com/mvp/uhubctl/issues/434), and I suspect your light bar is only a USB 2 device.

Have you checked if it works with USB 3 devices?

commit 3dbc1d9e (2024-08-31): Add Rosonway RSH-A107 (aka ikuai A107) as compatible device

Closes #572.