mvp/uhubctl #578: sysfs path not correctly constructed on some devices

bluecmd, 2024-08-03:
Hi,

On my device I can see that uhubctl is trying to access `/sys/bus/usb/devices/3:1.0/3-port1/disable` but the correct path for my USB controller is `/sys/bus/usb/devices/3-0:1.0/usb3-port1/disable`. 

lsusb output:
```
/:  Bus 001.Port 001: Dev 001, Class=root_hub, Driver=ehci-platform/2p, 480M
    |__ Port 001: Dev 002, If 0, Class=[unknown], Driver=usb-storage, 480M
/:  Bus 002.Port 001: Dev 001, Class=root_hub, Driver=ohci-platform/2p, 12M
/:  Bus 003.Port 001: Dev 001, Class=root_hub, Driver=xhci_hcd/2p, 480M
[..]
/:  Bus 004.Port 001: Dev 001, Class=root_hub, Driver=xhci_hcd/2p, 5000M
```

lspci output:
```
01:00.0 USB controller: Texas Instruments TUSB73x0 SuperSpeed USB 3.0 xHCI Host Controller (rev 02)
```

Files in `/sys/bus/usb/devices/`:
```
drwxr-xr-x    2 root     root           0 Jan  1  1970 .
drwxr-xr-x    4 root     root           0 Jan  1  1970 ..
lrwxrwxrwx    1 root     root           0 Jan  1  1970 1-0:1.0 -> ../../../devices/platform/soc@0/118006f000000.uctl/16f0000000000.ehci/usb1/1-0:1.0
lrwxrwxrwx    1 root     root           0 Jan  1  1970 1-1 -> ../../../devices/platform/soc@0/118006f000000.uctl/16f0000000000.ehci/usb1/1-1
lrwxrwxrwx    1 root     root           0 Jan  1  1970 1-1:1.0 -> ../../../devices/platform/soc@0/118006f000000.uctl/16f0000000000.ehci/usb1/1-1/1-1:1.0
lrwxrwxrwx    1 root     root           0 Jan  1  1970 2-0:1.0 -> ../../../devices/platform/soc@0/118006f000000.uctl/16f0000000400.ohci/usb2/2-0:1.0
lrwxrwxrwx    1 root     root           0 Jan  1  1970 3-0:1.0 -> ../../../devices/pci0000:01/0000:01:00.0/usb3/3-0:1.0
lrwxrwxrwx    1 root     root           0 Jan  1  1970 3-1 -> ../../../devices/pci0000:01/0000:01:00.0/usb3/3-1
lrwxrwxrwx    1 root     root           0 Jan  1  1970 3-1:1.0 -> ../../../devices/pci0000:01/0000:01:00.0/usb3/3-1/3-1:1.0
lrwxrwxrwx    1 root     root           0 Jan  1  1970 3-1:1.10 -> ../../../devices/pci0000:01/0000:01:00.0/usb3/3-1/3-1:1.10
lrwxrwxrwx    1 root     root           0 Jan  1  1970 3-1:1.11 -> ../../../devices/pci0000:01/0000:01:00.0/usb3/3-1/3-1:1.11
lrwxrwxrwx    1 root     root           0 Jan  1  1970 3-1:1.2 -> ../../../devices/pci0000:01/0000:01:00.0/usb3/3-1/3-1:1.2
lrwxrwxrwx    1 root     root           0 Jan  1  1970 3-1:1.3 -> ../../../devices/pci0000:01/0000:01:00.0/usb3/3-1/3-1:1.3
lrwxrwxrwx    1 root     root           0 Jan  1  1970 3-1:1.8 -> ../../../devices/pci0000:01/0000:01:00.0/usb3/3-1/3-1:1.8
lrwxrwxrwx    1 root     root           0 Aug  3 20:14 3-2 -> ../../../devices/pci0000:01/0000:01:00.0/usb3/3-2
lrwxrwxrwx    1 root     root           0 Aug  3 20:14 3-2:1.0 -> ../../../devices/pci0000:01/0000:01:00.0/usb3/3-2/3-2:1.0
lrwxrwxrwx    1 root     root           0 Jan  1  1970 4-0:1.0 -> ../../../devices/pci0000:01/0000:01:00.0/usb4/4-0:1.0
lrwxrwxrwx    1 root     root           0 Jan  1  1970 usb1 -> ../../../devices/platform/soc@0/118006f000000.uctl/16f0000000000.ehci/usb1
lrwxrwxrwx    1 root     root           0 Jan  1  1970 usb2 -> ../../../devices/platform/soc@0/118006f000000.uctl/16f0000000400.ohci/usb2
lrwxrwxrwx    1 root     root           0 Jan  1  1970 usb3 -> ../../../devices/pci0000:01/0000:01:00.0/usb3
lrwxrwxrwx    1 root     root           0 Jan  1  1970 usb4 -> ../../../devices/pci0000:01/0000:01:00.0/usb4
```

Ping @hnez who wrote the original sysfs implementation if my research is correct.

mvp, 2024-08-05:
Current code constructs path using:
```
    snprintf(disable_path, PATH_MAX,
        "/sys/bus/usb/devices/%s:%d.0/%s-port%i/disable",
        hub->location, configuration, hub->location, port
    );
```
For your case it emits this path: `/sys/bus/usb/devices/3:1.0/3-port1/disable`
when it should be `/sys/bus/usb/devices/3-0:1.0/3-port1/disable`.

Search kernel for naming conventions, I see both `<busnum>-<devnum>` used and [`<busnum>-<port>.<port>`](https://github.com/torvalds/linux/blob/de9c2c66ad8e/Documentation/ABI/stable/sysfs-bus-usb#L63) used. This commit also notes this discrepancy https://github.com/torvalds/linux/commit/159b4e3385d.

I guess we need deeper investigation into this to always get correct sysfs usb path...

bluecmd, 2024-08-05:
It turns out this is the case on my laptop as well, running 6.9.12 (Fedora 40). **I believe this is a special case for root ports** as opposed to external hubs.

```
sudo strace -etrace=file -f uhubctl -l 3 -p 7 -f -a off 2>&1 | grep disable
[pid 10936] openat(AT_FDCWD, "/sys/bus/usb/devices/3:1.0/3-port7/disable", O_WRONLY) = -1 ENOENT (No such file or directory)
```

The correct file on my laptop is `/sys/bus/usb/devices/3-0:1.0/usb3-port7/disable`.

Looking how the naming is constructed for normal USB hubs and ports in the Linux kernel, it seems like this case is what the kernel seems to do normally; [usb_hub_create_port_device](https://github.com/torvalds/linux/blob/de9c2c66ad8e787abec7c9d7eff4f8c3cdd28aed/drivers/usb/core/port.c#L763-L764)

Notably look at this: [usb_alloc_dev](https://github.com/torvalds/linux/blob/de9c2c66ad8e787abec7c9d7eff4f8c3cdd28aed/drivers/usb/core/usb.c#L682-L717)

Here the kernel seems to call the root ports `usb%d` and non-root ports just `%d`. This would explain the discrepancy.
Indeed, if I attach a USB 2.0 external hub things seems to work:

```
sudo strace -etrace=file -f uhubctl -l 3-1 -p 1 -f -a off 2>&1 | grep disable
[pid 11935] openat(AT_FDCWD, "/sys/bus/usb/devices/3-1:1.0/3-1-port1/disable", O_WRONLY) = 8
```

I suggest the following patch:

```patch
diff --git a/uhubctl.c b/uhubctl.c
index 6833224..3025162 100644
--- a/uhubctl.c
+++ b/uhubctl.c
@@ -574,10 +574,17 @@ static int set_port_status_linux(struct libusb_device_handle *devh, struct hub_i
      * The "disable" sysfs interface is available only starting with kernel version 6.0.
      * For earlier kernel versions the open() call will fail and we fall back to using libusb.
      */
-    snprintf(disable_path, PATH_MAX,
-        "/sys/bus/usb/devices/%s:%d.0/%s-port%i/disable",
-        hub->location, configuration, hub->location, port
-    );
+    if (hub->pn_len == 0) {
+      snprintf(disable_path, PATH_MAX,
+          "/sys/bus/usb/devices/%s-0:%d.0/usb%s-port%i/disable",
+          hub->location, configuration, hub->location, port
+      );
+    } else {
+      snprintf(disable_path, PATH_MAX,
+          "/sys/bus/usb/devices/%s:%d.0/%s-port%i/disable",
+          hub->location, configuration, hub->location, port
+      );
+    }
 
     int disable_fd = open(disable_path, O_WRONLY);
     if (disable_fd >= 0) {
```

A simple test on my laptop:

```
sudo strace -etrace=file -f ./uhubctl -l 3 -p 7 -f -a cycle 2>&1 | grep disable
[pid 13975] openat(AT_FDCWD, "/sys/bus/usb/devices/3-0:1.0/usb3-port7/disable", O_WRONLY) = 8
[pid 13975] openat(AT_FDCWD, "/sys/bus/usb/devices/3-0:1.0/usb3-port7/disable", O_WRONLY) = 8

sudo strace -etrace=file -f ./uhubctl -l 3-1 -p 2 -f -a cycle 2>&1 | grep disable
[pid 14045] openat(AT_FDCWD, "/sys/bus/usb/devices/3-1:1.0/3-1-port2/disable", O_WRONLY) = 8
[pid 14045] openat(AT_FDCWD, "/sys/bus/usb/devices/3-1:1.0/3-1-port2/disable", O_WRONLY) = 8
```

mvp, 2024-08-05:
Thank you @bluecmd, that patch looks good!
Let me know if you want to raise PR with this patch - if not, I can apply this fix myself.

PR #579 (2024-08-05): Construct root USB device sysfs path
Fixes #578. See issue for discussion.
commit 2df46818 (2024-08-05): Construct root USB device sysfs path

Fixes #578. See issue for discussion.
closed 2024-08-05 by mvp