mvp/uhubctl #584: Don't work report - Buffalo BSH4A05U3BK

suica-me, 2024-09-01:
Hello, thank you so much for your work!

I want to report that Buffalo BSH4A05U3BK maybe do not have a good power switching.
I try to use this to control my fan, but the fan just keeping work when I power off the hub.

---

The output of `uhubctl` and voltage test.
```
$ uhubctl          
Current status for hub 3-2 [05e3:0610 GenesysLogic USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
Current status for hub 2-2 [05e3:0617 GenesysLogic USB3.0 Hub, USB 3.00, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
$ uhubctl -a 0
$ uhubctl     
Current status for hub 3-2 [05e3:0610 GenesysLogic USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 1: 0000 off
  Port 2: 0000 off
  Port 3: 0000 off
  Port 4: 0000 off
Current status for hub 2-2 [05e3:0617 GenesysLogic USB3.0 Hub, USB 3.00, 4 ports, ppps]
  Port 1: 00a0 off
  Port 2: 00a0 off
  Port 3: 00a0 off
  Port 4: 00a0 off
```
[image]

You can see the hub still have a 5V voltage.

mvp, 2024-09-01:
Thanks for your report! I guess we need to remove this hub from supported list 😢.
I'm trying really hard to only add hubs that truly support vbus off, but unfortunately, people who report working hubs are not always doing comprehensive vbus tests...
To the best of my knowledge, number of such hubs in supported list should be very low.

closed 2024-09-01 by mvp

suica-me, 2024-09-01:
Thank you for your reply!
I totally agree, but it really isn't always easy for everyone to do an vbus test.
Thanks again for you work! I'll keep looking for a hub that works 😂.

mvp, 2024-09-01:
Removed from supported list in https://github.com/mvp/uhubctl/commit/bfc11e6b.

commit bfc11e6b (2024-09-01): Remove Buffalo BSH4A05U3BK from supported list

Buffalo BSH4A05U3BK does not seem to support VBUS off, removing from the list.

Closes #584.