mvp/uhubctl #585: Skull & Co. CoreDrive

TobiX, 2024-09-08:
The (old?) Skull & Co. CoreDrive (which is the base of the [Jumpgate](https://skullnco.com/collections/nintendo/products/jumpgate?variant=39442715279433) & [SteamDock](https://skullnco.com/collections/steam-deck/products/steamdock)) does support per-port power switching.

I have a CoreDrive with 2 USB 2.0 A ports & an SD card reader (while the current versions have either 1 USB 2.0 A port or 3 USB 3.0 A ports), so current versions might not be the same hardware at all.

```
Current status for hub 8-1 [2109:0817 VIA Labs, Inc. USB3.0 Hub, USB 3.10, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 0203 power 5gbps U0 enable connect [058f:8461 Generic Mass Storage Device 058F84618461]
  Port 3: 0203 power 5gbps U0 enable connect [05e3:0749 Generic USB3.0 Card Reader 000000001539]
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 7-1 [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
```

- The USB-A ports are Port 2 (outer) and Port 1 (between the HDMI port and the other USB port)
- Port 3 is the internal SD card reader, and port switching does disable the device, but does not cut power to the SD card
- Port 4 is not the upstream USB-C port :laughing: (probably not connected)

mvp, 2024-09-12:
Do ports 1 and 2 support true power off? If not, I don't want to add it to the table.

TobiX, 2024-09-13:
They do not, my testing was flawed. Sorry for the noise.

closed 2024-09-13 by TobiX