mvp/uhubctl #586: Build error: `error: expected declaration specifiers or '...' before '__builtin_object_size'`

barracuda156, 2024-09-11:
Build fails on 10.6, both with the old Xcode gcc and the modern gcc14.

gcc-4.2:
```
--->  Building uhubctl
Executing:  cd "/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_sysutils_uhubctl/uhubctl/work/uhubctl-2.5.0" && /usr/bin/make -j6 -w uhubctl prefix=/opt/local CC="/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_sysutils_uhubctl/uhubctl/work/compwrap/cc/usr/bin/gcc-4.2" CXX="/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_sysutils_uhubctl/uhubctl/work/compwrap/cxx/usr/bin/g++-4.2" OBJC="/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_sysutils_uhubctl/uhubctl/work/compwrap/objc/usr/bin/gcc-4.2" OBJCXX="/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_sysutils_uhubctl/uhubctl/work/compwrap/objcxx/usr/bin/g++-4.2" INSTALL="/usr/bin/install -c" 
fatal: not a git repository (or any of the parent directories): .git
make: Entering directory `/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_sysutils_uhubctl/uhubctl/work/uhubctl-2.5.0'
/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_sysutils_uhubctl/uhubctl/work/compwrap/cc/usr/bin/gcc-4.2 -I/opt/local/include -Os -arch ppc -Wall -Wextra -std=c99 -pedantic -DPROGRAM_VERSION=\"2.5.0\" -I/opt/local/include/libusb-1.0 uhubctl.c -o uhubctl -L/opt/local/lib -Wl,-headerpad_max_install_names -arch ppc -L/opt/local/lib -lusb-1.0
uhubctl.c:39: error: expected declaration specifiers or ‘...’ before numeric constant
uhubctl.c:39: error: expected declaration specifiers or ‘...’ before ‘__builtin_object_size’
uhubctl.c:39: warning: conflicting types for built-in function ‘__builtin___snprintf_chk’
uhubctl.c: In function ‘get_hub_info’:
uhubctl.c:427: warning: too many arguments for format
uhubctl.c:436: warning: too many arguments for format
uhubctl.c:441: warning: too many arguments for format
uhubctl.c: In function ‘get_device_description’:
uhubctl.c:698: warning: too many arguments for format
uhubctl.c:704: warning: too many arguments for format
uhubctl.c: In function ‘main’:
uhubctl.c:1049: warning: too many arguments for format
uhubctl.c:1055: warning: too many arguments for format
uhubctl.c:1058: warning: too many arguments for format
make: *** [uhubctl] Error 1
```

gcc-14:
```
--->  Building uhubctl
Executing:  cd "/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_sysutils_uhubctl/uhubctl/work/uhubctl-2.5.0" && /usr/bin/make -j6 -w uhubctl prefix=/opt/local CC="/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_sysutils_uhubctl/uhubctl/work/compwrap/cc/opt/local/bin/gcc-mp-14" CXX="/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_sysutils_uhubctl/uhubctl/work/compwrap/cxx/opt/local/bin/g++-mp-14" OBJC="/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_sysutils_uhubctl/uhubctl/work/compwrap/objc/opt/local/bin/gcc-mp-14" OBJCXX="/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_sysutils_uhubctl/uhubctl/work/compwrap/objcxx/opt/local/bin/g++-mp-14" FC="/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_sysutils_uhubctl/uhubctl/work/compwrap/fc/opt/local/bin/gfortran-mp-14" F77="/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_sysutils_uhubctl/uhubctl/work/compwrap/f77/opt/local/bin/gfortran-mp-14" F90="/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_sysutils_uhubctl/uhubctl/work/compwrap/f90/opt/local/bin/gfortran-mp-14" INSTALL="/usr/bin/install -c" 
fatal: not a git repository (or any of the parent directories): .git
make: Entering directory `/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_sysutils_uhubctl/uhubctl/work/uhubctl-2.5.0'
/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_sysutils_uhubctl/uhubctl/work/compwrap/cc/opt/local/bin/gcc-mp-14 -I/opt/local/include -Os -arch ppc -Wall -Wextra -std=c99 -pedantic -DPROGRAM_VERSION=\"2.5.0\" -I/opt/local/include/libusb-1.0 uhubctl.c -o uhubctl -L/opt/local/lib -Wl,-headerpad_max_install_names -arch ppc -L/opt/local/lib -lusb-1.0
In file included from /usr/include/stdio.h:431,
                 from uhubctl.c:14:
uhubctl.c:39:5: error: expected declaration specifiers or '...' before numeric constant
   39 | int snprintf(char * __restrict __str, size_t __size, const char * __restrict __format, ...) __printflike(3, 4);
      |     ^~~~~~~~
In file included from /usr/include/secure/_stdio.h:31:
uhubctl.c:39:5: error: expected declaration specifiers or '...' before '__builtin_object_size'
   39 | int snprintf(char * __restrict __str, size_t __size, const char * __restrict __format, ...) __printflike(3, 4);
      |     ^~~~~~~~
make: *** [uhubctl] Error 1
```

mvp, 2024-09-11:
It builds fine in brew, always has been. Must be something wrong with macports? Or with your local setup?

barracuda156, 2024-09-11:
@mvp Homebrew does not even work on earlier than a few recent macOS versions, so they cannot test there. The issue is likely caused by a slightly different declaration in macOS SDK. Those were not kept constant always, some did change over time.

mvp, 2024-09-11:
I remember building uhubctl on MacOS 10.10 and later, never were any issues. 10.6 does seem extremely ancient indeed - it was released 15 years ago after all. I would not be surprised if libusb didn't support it either.

barracuda156, 2024-09-11:
`libusb` builds fine, at least quite a number of its dependents build fine too.
I can try figuring out what goes wrong. Failures with gcc-4.2 are common, but gcc-14 should be able to build any correct code.

ryandesign, 2024-09-12:
Looks like you are declaring `snprintf` if on macOS or FreeBSD:

https://github.com/mvp/uhubctl/blob/bfc11e6b05ffcdf92eef24e7f728ac5282645274/uhubctl.c#L38-L40

and per the error message this seems to be conflicting with the system's declaration of `snprintf`. @barracuda156, does it build if you remove those lines?

This was added in 8e667cde10fe9d5aece33a8fadba453315cc2d03 where the commit message includes the line:

> use snprintf (which is weird on Mac)

without explaining what's meant by that. My initial reaction is that you should not be declaring system functions; the system should be the one to do so in its system headers. Your assertion that `snprintf` is not available in pure C mode seems unlikely. Instead, I see that you are defining `_XOPEN_SOURCE`:

https://github.com/mvp/uhubctl/blob/bfc11e6b05ffcdf92eef24e7f728ac5282645274/uhubctl.c#L12

When you do not define `_XOPEN_SOURCE` all functions are available but when you define it, it limits what functions are available. `500` may be sufficient to make `snprintf` available on Linux but perhaps a higher value like `600` is needed to make it appear on BSD-derived systems like macOS and FreeBSD; I've definitely encountered this with other string functions; for example see https://github.com/jordansissel/xdotool/issues/313.

mvp, 2024-09-12:
It would work if I used sprintf, but it is unsafe from security perspective. I wanted snprintf. However, on Darwin or FreeBSD, snprintf is only available for C++ compiler. I didn't want to make uhubctl C++ program either.
So easiest way to reconcile is to use 
```
 int snprintf(char * __restrict __str, size_t __size, const char * __restrict __format, ...) __printflike(3, 4); 
```
which is what Darwin/FreeBSD are using.

If you can find better way to satisfy all 3 conditions, I would appreciate it. That is: important snprintf, keep it C-only, and avoid conflicts on older systems.

ryandesign, 2024-09-12:
It is a C function; it is of course available to the C compiler. My suggested untested fix is as I already explained:

```diff
--- uhubctl.c.orig	2024-09-11 21:07:35.000000000 -0500
+++ uhubctl.c	2024-09-11 21:07:52.000000000 -0500
@@ -9,7 +9,7 @@
  *
  */
 
-#define _XOPEN_SOURCE 500
+#define _XOPEN_SOURCE 600
 
 #include <stdio.h>
 #include <stdlib.h>
@@ -35,10 +35,6 @@
 #include <libusb-1.0/libusb.h>
 #endif
 
-#if defined(__APPLE__) || defined(__FreeBSD__) /* snprintf is not available in pure C mode */
-int snprintf(char * __restrict __str, size_t __size, const char * __restrict __format, ...) __printflike(3, 4);
-#endif
-
 #if !defined(LIBUSB_API_VERSION) || (LIBUSB_API_VERSION <= 0x01000103)
 #define LIBUSB_DT_SUPERSPEED_HUB 0x2a
 #endif
```

mvp, 2024-09-12:
```
-#define _XOPEN_SOURCE 500
+#define _XOPEN_SOURCE 600
```
works for me. Just need to test that it actually works for most platforms.

@barracuda156 , can you test this on your very old Mac? Define XOPEN_SOURCE 600, delete snprintf override. Thanks!

barracuda156, 2024-09-12:
@ryandesign Ryan, thank you for helping here.

@mvp Sure, I will test this, though possibly tomorrow. I can try in Rosetta right now though.

barracuda156, 2024-09-12:
Ok, so I can reproduce the failure on 10.6.8:
```
--->  Building uhubctl
Executing:  cd "/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_sysutils_uhubctl/uhubctl/work/uhubctl-2.5.0" && /usr/bin/make -j6 -w uhubctl prefix=/opt/local CC="/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_sysutils_uhubctl/uhubctl/work/compwrap/cc/usr/bin/gcc-4.2" CXX="/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_sysutils_uhubctl/uhubctl/work/compwrap/cxx/usr/bin/g++-4.2" OBJC="/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_sysutils_uhubctl/uhubctl/work/compwrap/objc/usr/bin/gcc-4.2" OBJCXX="/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_sysutils_uhubctl/uhubctl/work/compwrap/objcxx/usr/bin/g++-4.2" INSTALL="/opt/x86_64/bin/ginstall -c" 
fatal: not a git repository (or any of the parent directories): .git
make: Entering directory `/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_sysutils_uhubctl/uhubctl/work/uhubctl-2.5.0'
/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_sysutils_uhubctl/uhubctl/work/compwrap/cc/usr/bin/gcc-4.2 -I/opt/local/include -Os -arch ppc -Wall -Wextra -std=c99 -pedantic -DPROGRAM_VERSION=\"2.5.0\" uhubctl.c -o uhubctl -L/opt/local/lib -Wl,-headerpad_max_install_names -arch ppc -lusb-1.0
uhubctl.c:39: error: expected declaration specifiers or ‘...’ before numeric constant
uhubctl.c:39: error: expected declaration specifiers or ‘...’ before ‘__builtin_object_size’
uhubctl.c:39: warning: conflicting types for built-in function ‘__builtin___snprintf_chk’
uhubctl.c: In function ‘get_hub_info’:
uhubctl.c:427: warning: too many arguments for format
uhubctl.c:436: warning: too many arguments for format
uhubctl.c:441: warning: too many arguments for format
uhubctl.c: In function ‘get_device_description’:
uhubctl.c:698: warning: too many arguments for format
uhubctl.c:704: warning: too many arguments for format
uhubctl.c: In function ‘main’:
uhubctl.c:1049: warning: too many arguments for format
uhubctl.c:1055: warning: too many arguments for format
uhubctl.c:1058: warning: too many arguments for format
make: *** [uhubctl] Error 1
make: Leaving directory `/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_sysutils_uhubctl/uhubctl/work/uhubctl-2.5.0'
Command failed:  cd "/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_sysutils_uhubctl/uhubctl/work/uhubctl-2.5.0" && /usr/bin/make -j6 -w uhubctl prefix=/opt/local CC="/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_sysutils_uhubctl/uhubctl/work/compwrap/cc/usr/bin/gcc-4.2" CXX="/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_sysutils_uhubctl/uhubctl/work/compwrap/cxx/usr/bin/g++-4.2" OBJC="/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_sysutils_uhubctl/uhubctl/work/compwrap/objc/usr/bin/gcc-4.2" OBJCXX="/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_sysutils_uhubctl/uhubctl/work/compwrap/objcxx/usr/bin/g++-4.2" INSTALL="/opt/x86_64/bin/ginstall -c" 
Exit code: 2
Error: Failed to build uhubctl: command execution failed
```

And Ryan’s suggestion works. I just edited the file and resumed the build, it completed:
```
Error: Processing of port uhubctl failed
macmini:~ svacchanda$ sudo port -v -n build uhubctl
--->  Computing dependencies for uhubctl.
--->  Building uhubctl
Executing:  cd "/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_sysutils_uhubctl/uhubctl/work/uhubctl-2.5.0" && /usr/bin/make -j6 -w uhubctl prefix=/opt/local CC="/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_sysutils_uhubctl/uhubctl/work/compwrap/cc/usr/bin/gcc-4.2" CXX="/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_sysutils_uhubctl/uhubctl/work/compwrap/cxx/usr/bin/g++-4.2" OBJC="/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_sysutils_uhubctl/uhubctl/work/compwrap/objc/usr/bin/gcc-4.2" OBJCXX="/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_sysutils_uhubctl/uhubctl/work/compwrap/objcxx/usr/bin/g++-4.2" INSTALL="/opt/x86_64/bin/ginstall -c" 
fatal: not a git repository (or any of the parent directories): .git
make: Entering directory `/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_sysutils_uhubctl/uhubctl/work/uhubctl-2.5.0'
/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_sysutils_uhubctl/uhubctl/work/compwrap/cc/usr/bin/gcc-4.2 -I/opt/local/include -Os -arch ppc -Wall -Wextra -std=c99 -pedantic -DPROGRAM_VERSION=\"2.5.0\" uhubctl.c -o uhubctl -L/opt/local/lib -Wl,-headerpad_max_install_names -arch ppc -lusb-1.0
make: Leaving directory `/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_sysutils_uhubctl/uhubctl/work/uhubctl-2.5.0'
```

mvp, 2024-09-12:
Fixed in 153c5da2. Thanks @ryandesign for suggestion!

commit 153c5da2 (2024-09-12): Fixed snprintf override which breaks on old MacOS

`snprintf` is available in Mac/FreeBSD headers only if _XOPEN_SOURCE is 600 or higher.
This closes #586.
closed 2024-09-12 by mvp