mvp/uhubctl #587: running Ubuntu 24.04 on Raspberry Pi 5

sb-fwr, 2024-09-11:
Hi,
I tried using this lib on Ubuntu 24.04 on a Pi 5. i can deactivate the USB2 ports, but the USB3 Ports stay active. I tried using raspbian, there it works fine. Any ideas what the problem might be?

mvp, 2024-09-12:
Please provide more information - Linux kernel version, uhubctl version, uhubctl output.
Note that uhubctl version 2.6.0 has a fix for issue #578 which affects operation on kernel 6.x for root USB3 hubs. If you installed it from Ubuntu packages, it would be version 2.5.0 and is most likely affected by this bug. I recommend installing 2.6.0 or compiling it from source.

sb-fwr, 2024-09-13:
Sorry, that was a bit vague.
I am using 'version: 2.6.0-2-gbfc11e6b', the Ubuntu 24.04 is the one raspberry-imager uses. Linux kernel '6.8.0-1010-raspi'. 
I ran your Pi5 example as root and got: 'root@stereocampi:/home/pi/uhubctl# uhubctl -l 1 -a 0
uhubctl -l 3 -a 0
Current status for hub 1 [1d6b:0002 Linux 6.8.0-1010-raspi dwc2_hsotg DWC OTG Controller 1000480000.usb, USB 2.00, 1 ports, ppps]
  Port 1: 0000 off
Sent power off request
New status for hub 1 [1d6b:0002 Linux 6.8.0-1010-raspi dwc2_hsotg DWC OTG Controller 1000480000.usb, USB 2.00, 1 ports, ppps]
  Port 1: 0000 off
No compatible devices detected at location 3!
Run with -h to get usage info.'

mvp, 2024-09-13:
Can you please show output of `sudo uhubctl `?

There is a chance that you have bus numbers assigned differently.

Do you remember which kernel version you had on your raspbian when it was working?

Also, was it on the same physical hardware?

sb-fwr, 2024-09-16:
the raspbian kernel version was 6.6.47.
on raspbian :sudo uhubctl
Current status for hub 4 [1d6b:0003 Linux 6.6.47+rpt-rpi-2712 xhci-hcd xHCI Host Controller xhci-hcd.1, USB 3.00, 1 ports, ppps]
  Port 1: 0203 power 5gbps U0 enable connect [2560:c128 e-con systems See3CAM_24CUG 1630D309]
Current status for hub 3-2 [2109:2812 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0103 power enable connect [24f0:0140 Metadot - Das Keyboard Das Keyboard]
Current status for hub 3 [1d6b:0002 Linux 6.6.47+rpt-rpi-2712 xhci-hcd xHCI Host Controller xhci-hcd.1, USB 2.00, 2 ports, ppps]
  Port 1: 0100 power
  Port 2: 0503 power highspeed enable connect [2109:2812 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
Current status for hub 2 [1d6b:0003 Linux 6.6.47+rpt-rpi-2712 xhci-hcd xHCI Host Controller xhci-hcd.0, USB 3.00, 1 ports, ppps]
  Port 1: 0203 power 5gbps U0 enable connect [2560:c128 e-con systems See3CAM_24CUG 1B0CD309]
Current status for hub 1 [1d6b:0002 Linux 6.6.47+rpt-rpi-2712 xhci-hcd xHCI Host Controller xhci-hcd.0, USB 2.00, 2 ports, ppps]
  Port 1: 0100 power
  Port 2: 0103 power enable connect [046d:c52b Logitech USB Receiver]

on Ubuntu 24.04:sudo uhubctl
Current status for hub 1 [1d6b:0002 Linux 6.8.0-1010-raspi dwc2_hsotg DWC OTG Controller 1000480000.usb, USB 2.00, 1 ports, ppps]
  Port 1: 0000 off
Current status for hub 5 [1d6b:0003 Linux 6.8.0-1010-raspi xhci-hcd xHCI Host Controller xhci-hcd.1, USB 3.00, 1 ports, ppps]
  Port 1: 0203 power 5gbps U0 enable connect [2560:c128 e-con systems See3CAM_24CUG 1630D309]
Current status for hub 4-2 [2109:2812 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0103 power enable connect [24f0:0140 Metadot - Das Keyboard Das Keyboard]
Current status for hub 4 [1d6b:0002 Linux 6.8.0-1010-raspi xhci-hcd xHCI Host Controller xhci-hcd.1, USB 2.00, 2 ports, ppps]
  Port 1: 0100 power
  Port 2: 0503 power highspeed enable connect [2109:2812 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
Current status for hub 3 [1d6b:0003 Linux 6.8.0-1010-raspi xhci-hcd xHCI Host Controller xhci-hcd.0, USB 3.00, 1 ports, ppps]
  Port 1: 0203 power 5gbps U0 enable connect [2560:c128 e-con systems See3CAM_24CUG 1B0CD309]
Current status for hub 2 [1d6b:0002 Linux 6.8.0-1010-raspi xhci-hcd xHCI Host Controller xhci-hcd.0, USB 2.00, 2 ports, ppps]
  Port 1: 0100 power
  Port 2: 0103 power enable connect [046d:c52b Logitech USB Receiver]

mvp, 2024-09-16:
Thanks, you have confirmed my suspicions.

This otg controller detection was added in more recent kernel, and this has shifted hub numbers by 1:
```
hub 1 [1d6b:0002 Linux 6.8.0-1010-raspi dwc2_hsotg DWC OTG Controller 1000480000.usb, USB 2.00, 1 ports, ppps]
```
Unfortunately, in my previous hack to support Pi 5 I have hardcoded bus numbers 1 and 3, which apparently not always true.

I will release a fix for this soon, as well as update documentation about this possible discrepancy.

In meantime, you can turn off power using -e workaround:
```
sudo uhubctl -e -a 0 -l 2
sudo uhubctl -e -a 0 -l 3
sudo uhubctl -e -a 0 -l 4
sudo uhubctl -e -a 0 -l 5
```
For -l parameter, use 4 bus numbers for all onboard hubs except for otg controller. For older kernels, they are 1,2,3,4.

mvp, 2024-09-17:
This issue should be fixed by e0d1c34. Please confirm @sb-fwr . Thanks!

closed 2024-09-17 by mvp

sb-fwr, 2024-09-17:
@mvp  Works like a charm. Thank you very much.

ZeChArtiahSaher, 2024-11-10:
for some reason if you enable any individual one it turns all of them on or at least powers them on

commit e0d1c34e (2024-09-17): Improve support for Raspberry Pi 4B and 5:

* Detect if we are running on Raspberry 4B or 5
* Add RPi 4B and 5 conditionals for RPi hacks
* Relax some conditions which may vary depending on RPi hardware and/or OS
  (in particular Ubuntu hub numbering differs from Raspbian).
* For RPi5, changed README to suggest using busses 2 and 4 instead of 1 and 3.
  This makes the same recipe work on both Raspbian and Ubuntu.

This fixes #587.