mvp/uhubctl #590: Amazon Basics 10-port Hub Ver Possibly Unsupported

alexpopatanasov, 2024-09-30:
Hey! I have the German version (EU connector) of the 10-port Amazon Basics Hub and have trouble making it work. Reported PPPS, but it switches only data off. Tried it with a USB fan and multiple other tests. 

`Current status for hub 2-4.4.4 [2109:0817 VIA Labs, Inc. USB3.0 Hub, USB 3.20, 4 ports, ppps]
  Port 1: 0203 power 5gbps U0 enable connect []
  Port 2: 0203 power 5gbps U0 enable connect []
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 2-4.4 [2109:0817 VIA Labs, Inc. USB3.0 Hub, USB 3.20, 4 ports, ppps]
  Port 1: 02e0 power 5gbps Polling
  Port 2: 0203 power 5gbps U0 enable connect []
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 0203 power 5gbps U0 enable connect []
Current status for hub 2-4 [2109:0817 VIA Labs, Inc. USB3.0 Hub, USB 3.20, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 0203 power 5gbps U0 enable connect []
Current status for hub 1-4.4.4 [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0503 power highspeed enable connect []
  Port 4: 0100 power
Current status for hub 1-4.4 [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0503 power highspeed enable connect [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
Current status for hub 1-4 [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0503 power highspeed enable connect [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
`

`Bus 001 Device 005: ID 2109:2817 VIA Labs, Inc. USB2.0 Hub             
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.10
  bDeviceClass            9 Hub
  bDeviceSubClass         0 [unknown]
  bDeviceProtocol         2 TT per port
  bMaxPacketSize0        64
  idVendor           0x2109 VIA Labs, Inc.
  idProduct          0x2817 USB2.0 Hub             
  bcdDevice           90.23
  iManufacturer           1 VIA Labs, Inc.         
  iProduct                2 USB2.0 Hub             
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0029
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower                0mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0 [unknown]
      bInterfaceProtocol      1 Single TT
[... 156 lines cut ...]
    bDescriptorType        16
    bDevCapabilityType      4
    bReserved               0
    ContainerID             {30eef35c-07d5-2549-b001-802d79434c30}
Hub Descriptor:
  bLength              12
  bDescriptorType      42
  nNbrPorts             4
  wHubCharacteristic 0x0009
    Per-port power switching
    Per-port overcurrent protection
  bPwrOn2PwrGood      175 * 2 milli seconds
  bHubContrCurrent      0 milli Ampere
  bHubDecLat          0.4 micro seconds
  wHubDelay          2292 nano seconds
  DeviceRemovable    0x10
 Hub Port Status:
   Port 1: 0000.02a0 5Gbps power Rx.Detect
   Port 2: 0000.02a0 5Gbps power Rx.Detect
   Port 3: 0000.02a0 5Gbps power Rx.Detect
   Port 4: 0000.0203 5Gbps power enable connect
Device Status:     0x000d
  Self Powered
  U1 Enabled
  U2 Enabled`

mvp, 2024-09-30:
Which version of uhubctl you are using? It is very suspicious that it shows `[]` empty brackets - it seems like very old version. Make sure you update to uhubctl 2.6.0, or at least 2.5.0.

alexpopatanasov, 2024-09-30:
No, I redacted the contents of it, they display correctly. It has 2 HDDs and 3 USB Flash Disk Drives.

mvp, 2024-09-30:
I was wrong, there are 10 port Amazon Basics hubs listed as compatible. Do you know what is the model/revision for yours?

alexpopatanasov, 2024-09-30:
¯\_(ツ)_/¯. Sadly, no idea. Box is pretty generic like the others. PID and VID are the same.

mvp, 2024-09-30:
Maybe you have Amazon link since you got it from Amazon?

alexpopatanasov, 2024-09-30:
https://www.amazon.de/-/en/dp/B076YRSWGW

mvp, 2024-10-09:
Sorry for late reply as I am traveling. Can you please tell which uhubctl version you are using? If it is not 2.5.0 or higher, please update to latest and post output of `sudo uhubctl`.

Also, it would be helpful if you can find model number, typically it is a sticker or barcode affixed to the back. Thanks!

mvp, 2024-11-17:
Closing by inactivity. Please reopen if you have anything to add.

closed 2024-11-17 by mvp