mvp/uhubctl #591: I-O DATA US3C-UERGB/H2 supported

programus, 2024-10-12:
I am looking for a solution to switch the power of usb port and found this wonderful tool!

I have none of usb hub in the supported list, so I tested this on the hub I have, [I-O DATA US3C-UERGB/H2](https://www.iodata.jp/product/av/graphicsboard/us3c-uergbh2/), which I get from my company many years ago. 

There is only one USB output port on this hub (others are LAN port, VGA port and HDMI port), so I tried to switch the power of this only port off and on. I found it is supported without any problem. (I am using mac-mini with macOS 15.0.1 installed. Installed uhubctl by homebrew. )

I-O Data is a Japanese company, and this hub is also mainly be used in Japan. And unfortunately, the home page seems also Japanese only. Well, I suppose this would not be a problem now since we have Google translation, ChatGPT and lots of other smart tools. 

So, I would like to report this support. 

Here is the log:

```
❯ uhubctl -h
uhubctl: utility to control USB port power for smart hubs.
Usage: uhubctl [options]
Without options, show status for all smart hubs.

Options [defaults in brackets]:
--action,   -a - action to off/on/cycle/toggle/flash (0/1/2/3/4) for affected ports.
--ports,    -p - ports to operate on    [all hub ports].
--location, -l - limit hub by location  [all smart hubs].
--level     -L - limit hub by location level (e.g. a-b.c is level 3).
--vendor,   -n - limit hub by vendor id [any] (partial ok).
--search,   -s - limit hub by attached device description.
--delay,    -d - delay for cycle/flash action [2 sec].
--repeat,   -r - repeat power off count [1] (some devices need it to turn off).
--exact,    -e - exact location (no USB3 duality handling).
--force,    -f - force operation even on unsupported hubs.
--nodesc,   -N - do not query device description (helpful for unresponsive devices).
--reset,    -R - reset hub after each power-on action, causing all devices to reassociate.
--wait,     -w - wait before repeat power off [20 ms].
--version,  -v - print program version.
--help,     -h - print this text.

Send bugs and requests to: https://github.com/mvp/uhubctl
version: 2.6.0
❯ uhubctl
Current status for hub 1-2 [04bb:0603 Cypress Semiconductor CY-HX3 HUB DCD364202405, USB 3.00, 2 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 0203 power 5gbps U0 enable connect [0b95:1790 ASIX Elec. Corp. AX88179 023476C5D3F7C4]
Current status for hub 1-1 [04bb:0604 Cypress Semiconductor CY-HX3 HUB DCD364202405, USB 2.10, 3 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0103 power enable connect [04bb:0605 Cypress Semiconductor Billboard Device 102405]
❯ uhubctl -a off -p 1
Current status for hub 1-2 [04bb:0603 Cypress Semiconductor CY-HX3 HUB DCD364202405, USB 3.00, 2 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
Sent power off request
New status for hub 1-2 [04bb:0603 Cypress Semiconductor CY-HX3 HUB DCD364202405, USB 3.00, 2 ports, ppps]
  Port 1: 00a0 off
Current status for hub 1-1 [04bb:0604 Cypress Semiconductor CY-HX3 HUB DCD364202405, USB 2.10, 3 ports, ppps]
  Port 1: 0100 power
Sent power off request
New status for hub 1-1 [04bb:0604 Cypress Semiconductor CY-HX3 HUB DCD364202405, USB 2.10, 3 ports, ppps]
  Port 1: 0000 off
❯ uhubctl
Current status for hub 1-2 [04bb:0603 Cypress Semiconductor CY-HX3 HUB DCD364202405, USB 3.00, 2 ports, ppps]
  Port 1: 00a0 off
  Port 2: 0203 power 5gbps U0 enable connect [0b95:1790 ASIX Elec. Corp. AX88179 023476C5D3F7C4]
Current status for hub 1-1 [04bb:0604 Cypress Semiconductor CY-HX3 HUB DCD364202405, USB 2.10, 3 ports, ppps]
  Port 1: 0000 off
  Port 2: 0100 power
  Port 3: 0103 power enable connect [04bb:0605 Cypress Semiconductor Billboard Device 102405]
❯ uhubctl -a on -p 1
Current status for hub 1-2 [04bb:0603 Cypress Semiconductor CY-HX3 HUB DCD364202405, USB 3.00, 2 ports, ppps]
  Port 1: 00a0 off
Sent power on request
New status for hub 1-2 [04bb:0603 Cypress Semiconductor CY-HX3 HUB DCD364202405, USB 3.00, 2 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
Current status for hub 1-1 [04bb:0604 Cypress Semiconductor CY-HX3 HUB DCD364202405, USB 2.10, 3 ports, ppps]
  Port 1: 0000 off
Sent power on request
New status for hub 1-1 [04bb:0604 Cypress Semiconductor CY-HX3 HUB DCD364202405, USB 2.10, 3 ports, ppps]
  Port 1: 0100 power
❯ uhubctl
Current status for hub 1-2 [04bb:0603 Cypress Semiconductor CY-HX3 HUB DCD364202405, USB 3.00, 2 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 0203 power 5gbps U0 enable connect [0b95:1790 ASIX Elec. Corp. AX88179 023476C5D3F7C4]
Current status for hub 1-1 [04bb:0604 Cypress Semiconductor CY-HX3 HUB DCD364202405, USB 2.10, 3 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0103 power enable connect [04bb:0605 Cypress Semiconductor Billboard Device 102405]
 ```

But my problem is not solved yet because this hub does not provide enough power to turn my monitor on... I may have to buy one from the supported list. 😭

mvp, 2024-10-12:
Can you please do VBUS test? Check if connected phone stops charging, or USB can stops spinning?

mvp, 2024-11-14:
Closing by inactivity. If you can do VBUS test, I will add this as supported hub. Thanks!

closed 2024-11-14 by mvp

programus, 2024-11-23:
> Can you please do VBUS test? Check if connected phone stops charging, or USB fan stops spinning?

@mvp Sorry for responding so late. 

I had done the VBUS test while I reported this, and I also confirmed it again just now. 

I connected a USB battery charger and the LED lights are on if I turn the port on, while the lights are off if I turn the port off.