mvp/uhubctl #596: Does Waveshare USB-HUB-7U work?

SlySven, 2024-11-08:
Was checking out an on-line auction site and came across this unit: https://www.waveshare.com/usb-hub-7u.htm looking at the PWB picture it seems possible that this unit could have individual switching potential. I note that #541 was for a device (3x USB + 1x LAN ports) from the same company but that didn't do Per-Port (or any) Switching.

mvp, 2024-11-08:
I have no idea. If anyone can get it and test it, would be great.

SlySven, 2024-11-14:
Have sent a support ticket request to the manufacturer - am awaiting a reply...

SlySven, 2024-11-19:
I had a reply - and it was rather negative:
> Your request (181607) has been updated. To add additional comments, reply all to this email.
>
>Hitesh
>Nov 15, 2024, 05:42 UTC
>
>Hello
>
> 
>
> No it does not support software turn on off USB feature.
>
>> Visitor 577600369633
>> Nov 14, 2024, 18:32 UTC
>> I spotted your unit being advertised on EBay and was wondering, after looking over the PWB in the images, whether it physically supported "Per-Port Power Supply Switching" i.e. so that utilities such as the "uhubctrl" project could turn the power on/off on individual ports via standard (but often not implemented!) USB protocol commands? I'm trying to find out so as to be able to fill out: https://github.com/mvp/uhubctl/issues/596 - and possibly to buy one for my own Raspberry Pi "Home Automation" project {I had a RFID tag reader on my front door would like to be able to power it off once I've deadlocked the door and the electric lock I have fitted cannot be operated to open it}...
>>
>>I note that someone else has already determined that another product of yours (the "USB-HUB-3U-ETH") does NOT have that capability: https://github.com/mvp/uhubctl/pull/541 .

closed 2024-11-19 by SlySven