mvp/uhubctl #597: Is BIG7: 7-Port MTT USB Hub for Raspberry Pi (Rev 2) compatible with uhubctl?

AubinF, 2024-11-14:
Hi, thank you for maintaining uhubctl, I use it all the time!

I was wondering if [BIG7: 7-Port MTT USB Hub for Raspberry Pi (Rev 2)](https://www.uugear.com/product/7-port-usb-hub-for-raspberry-pi/) would work with uhubctl?

Thank you

mvp, 2024-11-14:
I have no idea. Most devices in supported table were reported by users like you. Please check and report! 🤣

mvp, 2024-11-14:
Actually, I looked deeper at specs, and I believe it will NOT work.
It was created by @uugear - same vendor who created MEGA4 (which is fully compatible with uhubctl).

Per their [documentation](https://www.uugear.com/product/7-port-usb-hub-for-raspberry-pi):
```
...the power of USB ports could be controlled by software or external switch (note: it is not Per-Port Power Switching (PPPS), please read the [user manual](http://www.uugear.com/doc/BIG7_Rev2_UserManual.pdf) to learn how it works).
```
Crucial part is `it is NOT Per-Port Power Switching`.
So answer is no.

closed 2024-11-14 by mvp

AubinF, 2024-11-15:
Ah, I was about to order one to run some tests when I saw your answer. That's a shame, I like MEGA4 but it would be really nice to have more channels. Thank you for checking!