mvp/uhubctl #598: RSHTECH RSH-A37S is supported (tested)

tstrus97, 2024-11-15:
link to device: https://www.amazon.de/dp/B088FFNBT3?ref=cm_sw_r_apan_dp_WVKSHZH3J82HY7BNDW71&ref_=cm_sw_r_apan_dp_WVKSHZH3J82HY7BNDW71&social_share=cm_sw_r_apan_dp_WVKSHZH3J82HY7BNDW71&starsLeft=1&skipTwisterOG=1&th=1

Here is the proof that it works. I tested with an USB camera, which successfully gets turned off/on. Note test camera checks if the camera on port 2 is online.

```
(hil) hil@hil-3:~$ uhubctl
Current status for hub 2-2.4 [0bda:0411 Generic USB3.2 Hub, USB 3.20, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 2-2 [0bda:0411 Generic USB3.2 Hub, USB 3.20, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 0263 power 5gbps U3 enable connect [0bda:0411 Generic USB3.2 Hub, USB 3.20, 4 ports, ppps]
Current status for hub 1-6.4 [0bda:5411 Generic USB2.1 Hub, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0503 power highspeed enable connect [05c6:901d QCOM QCS_KALAMAP-HDK _SN:2D7CD107 bda64eb2]
  Port 4: 0100 power
Current status for hub 1-6 [0bda:5411 Generic USB2.1 Hub, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0503 power highspeed enable connect [03e7:2485 Movidius Ltd. Movidius MyriadX 03e72485]
  Port 3: 0100 power
  Port 4: 0503 power highspeed enable connect [0bda:5411 Generic USB2.1 Hub, USB 2.10, 4 ports, ppps]

(hil) hil@hil-3:~$ sudo uhubctl -a off -l 2-2  -p 2
Current status for hub 2-2 [0bda:0411 Generic USB3.2 Hub, USB 3.20, 4 ports, ppps]
  Port 2: 02a0 power 5gbps Rx.Detect
Sent power off request
New status for hub 2-2 [0bda:0411 Generic USB3.2 Hub, USB 3.20, 4 ports, ppps]
  Port 2: 00a0 off
Current status for hub 1-6 [0bda:5411 Generic USB2.1 Hub, USB 2.10, 4 ports, ppps]
  Port 2: 0503 power highspeed enable connect [03e7:2485 Movidius Ltd. Movidius MyriadX 03e72485]
Sent power off request
New status for hub 1-6 [0bda:5411 Generic USB2.1 Hub, USB 2.10, 4 ports, ppps]
  Port 2: 0000 off

(hil) hil@hil-3:~$ 
(hil) hil@hil-3:~$ python testCamera.py 

(hil) hil@hil-3:~$ sudo uhubctl -a on -l 2-2  -p 2
Current status for hub 2-2 [0bda:0411 Generic USB3.2 Hub, USB 3.20, 4 ports, ppps]
  Port 2: 00a0 off
Sent power on request
New status for hub 2-2 [0bda:0411 Generic USB3.2 Hub, USB 3.20, 4 ports, ppps]
  Port 2: 02a0 power 5gbps Rx.Detect
Current status for hub 1-6 [0bda:5411 Generic USB2.1 Hub, USB 2.10, 4 ports, ppps]
  Port 2: 0000 off
Sent power on request
New status for hub 1-6 [0bda:5411 Generic USB2.1 Hub, USB 2.10, 4 ports, ppps]
  Port 2: 0101 power connect []

(hil) hil@hil-3:~$ python testCamera.py 
DeviceInfo(name=1.6.2, mxid=19443010A142F91200, X_LINK_UNBOOTED, X_LINK_USB_VSC, X_LINK_MYRIAD_X, X_LINK_SUCCESS)

```

tstrus97, 2024-11-15:
I also testbed RSHTECH RSH-A13 which does not actually work. The uhubctl does output all the info, but powering off ports does not actually work.

```
(hil) hil@hil-3:~$ uhubctl
Current status for hub 2-1.4.3 [2109:0822 VIA Labs, Inc. USB3.1 Hub 000000001, USB 3.20, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 2-1.4 [2109:0822 VIA Labs, Inc. USB3.1 Hub 000000001, USB 3.20, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 0263 power 5gbps U3 enable connect [2109:0822 VIA Labs, Inc. USB3.1 Hub 000000001, USB 3.20, 4 ports, ppps]
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 2-1.3 [2109:0822 VIA Labs, Inc. USB3.1 Hub 000000001, USB 3.20, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 2-1 [2109:0822 VIA Labs, Inc. USB3.1 Hub 000000001, USB 3.20, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 0263 power 5gbps U3 enable connect [2109:0822 VIA Labs, Inc. USB3.1 Hub 000000001, USB 3.20, 4 ports, ppps]
  Port 4: 0263 power 5gbps U3 enable connect [2109:0822 VIA Labs, Inc. USB3.1 Hub 000000001, USB 3.20, 4 ports, ppps]
Current status for hub 1-5.4.3 [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
Current status for hub 1-5.4 [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0507 power highspeed suspend enable connect [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
  Port 4: 0507 power highspeed suspend enable connect [03e7:2485 Movidius Ltd. Movidius MyriadX 03e72485]
Current status for hub 1-5.3 [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
Current status for hub 1-5 [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0507 power highspeed suspend enable connect [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
  Port 4: 0503 power highspeed enable connect [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
(hil) hil@hil-3:~$ 
```

mvp, 2024-11-15:
How did you test A13? Testing voltage or at least phone charging? I find it a bit suspicious that some devices have empty brackets - that was only possible in very old version of uhubctl. Which version are you using? uhubctl -v?

tstrus97, 2024-11-15:
I tested it the samme as above, with a camera. For these hubs you also have LED lights that reflect the power state, which on A13 did not change at  all, but for A37S it works flawlesly
```
(hil) hil@hil-3:~$ uhubctl -v
2.6.0-7-g2dbc6ce2

```

mvp, 2025-01-18:
Thank you for submission, added to the list

commit f4b0d2ad (2025-01-18): Add RSH-A37S as compatible device

 Closes #598.
closed 2025-01-18 by mvp