mvp/uhubctl #599: feature request: integration for home assistant os

MHx489, 2024-11-17:
I used this nice function long time at different RPi's and did some automation with JS scripts at ioBroker. Like turn on/off USB power for USB Fans if temperatures going up and down inside a rack.
I switched over to home assistant OS and couldnt find any solution to turn off the RPi 5 USB ports anymore, because it's running in docker and you can't much access the RPi system.
Maybe it's possible for you to develop some intergration for home assistent. I think many people would love it, because i found a lot of users with same issues at my research and buying some Sonoff Smart USB adapter like others for that little function isn't much the case for me.

mvp, 2024-11-17:
uhubctl focus is to provide a functionality of switching USB power off and on, and getting basic information about connected USB hubs and devices. Various smart home integrations (Home Assistant, HomeKit etc) are not in scope.

However, there are few other projects that utilized uhubctl to provide such integrations. I recommend reading this article (featured in list of notable projects) https://medium.com/@soukron/how-to-manage-remotely-a-cinema-lightbox-rpi-uhubctl-homebridge-a933e6e178a5.
I'm sure there are many different ways to do that integration.
Note that if you intend to make it work on Raspberry Pi 5, it has a behavior that makes it different from any previous Raspberry Pi versions - namely, you have to execute it twice for locations 2 and 4 to actually turn power off for all ports - check readme section for Raspberry Pi 5. In all previous versions, it was always possible to turn power off with just a single command.

closed 2024-11-17 by mvp

MHx489, 2024-12-09:
Thanks for your answer. I was playing around a bit. HomeAssistant/Raspi uses Alpine OS. I manged to shutdown the USB 2.0 ports of the Raspi 5 at hub 1 and hub 3. Unfortunately it doesnt work for the USB3 ports at hub 2 and 4, what i need to shutdown the power completely of the ports. (No compatible devices detected at location 2!) Any idea what could help?

following works:
sudo uhubctl -l 1 -a 0 
sudo uhubctl -l 3 -a 0

error:
sudo uhubctl -l 2 -a 0 
No compatible devices detected at location 2!
Run with -h to get usage info.
sudo uhubctl -l 4 -a 0 
No compatible devices detected at location 4!
Run with -h to get usage info.

homeassistant:/config# sudo uhubctl  
Current status for hub 2 [1d6b:0003 Linux 6.6.51-haos-raspi xhci-hcd xHCI Host Controller xhci-hcd.0, USB 3.00, 1 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
Current status for hub 4 [1d6b:0003 Linux 6.6.51-haos-raspi xhci-hcd xHCI Host Controller xhci-hcd.1, USB 3.00, 1 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
Current status for hub 1 [1d6b:0002 Linux 6.6.51-haos-raspi xhci-hcd xHCI Host Controller xhci-hcd.0, USB 2.00, 2 ports, ppps]
  Port 1: 0000 off
  Port 2: 0000 off
Current status for hub 3 [1d6b:0002 Linux 6.6.51-haos-raspi xhci-hcd xHCI Host Controller xhci-hcd.1, USB 2.00, 2 ports, ppps]
  Port 1: 0000 off
  Port 2: 0000 off

mvp, 2024-12-09:
Please read documentation carefully https://github.com/mvp/uhubctl#raspberry-pi-5
Note that Raspberry Pi 5 support is rather new and requires you building uhubctl from master branch.

To turn off VBUS power it has to be disabled across all onboard hubs and ports with:
```
uhubctl -l 2 -a 0
uhubctl -l 4 -a 0
```
To turn it back on:
```
uhubctl -l 2 -a 1
uhubctl -l 4 -a 1
```
Note that VBUS power goes down only if all ports are off - enabling any single port enables VBUS back for all 4 ports.