mvp/uhubctl #601: Another compatible hub

loupiote, 2024-11-30:
I found that this hub (not in the supported list, so i had to use --force) works with uhubctl:

StarTech model HB31C2A2CB

USB ids:
Manufacturer: 14B0
ProductId: 013F
Rev: 0003

Note: On the device itself, it says "Rev 2"
I only tested with this revision, but I assume other revisions will work too.

The Hub has 2 USB-A Ports and  2 USB-C ports. Only the USB-A ports are power-controlled with uhubctl

mvp, 2025-01-18:
Thanks for your submission, added it to the list.

closed 2025-01-18 by mvp

Montg0mery, 2025-03-28:
I've noticed that with this hub (and some others), USB 2 devices (or USB 3 devices connected via a USB 2 cable) power off correctly, but USB 3 devices do not (there are no errors, but like you say, the devices disappear and re-appear on the host but the power / LEDs on the device never drops).

I think any testing / adding of hubs to the list should include testing both USB 2 and USB 3 devices, so that any differences can be noted.

Montg0mery, 2025-03-28:
Actually I've got it working. The only way the power turns off to a USB 3 port is if every USB 2 and every USB 3 port is powered off. Then the USB  device successfully powers down. I guess that makes it a bit like the Raspberry Pi behaviour?

commit 7604f84c (2025-01-18): Add StarTech HB31C2A2CB as compatible device

Closes #601.