mvp/uhubctl #602: Sometimes getting sementation fault (core dump)

loupiote, 2024-12-02:
I observed uhubctl doing segmentation fault (core dump) several times on Ububtu, when plugging or unplugging a USB device (e.g. external SSD) on one of the other USB ports of my laptop (not on the hub uhubctl was interacting with), while uhubctl was running.

I think that when I plugged my other device, for a short time it reset other USB ports, including the one where my hub is plugged, so I think uhubctl ended up interacting with a hub that became offline for a short time..

Once I got a message that the SIGSEGV happened in a library (i think libusb.so ?)

Have you ever noticed that?

Basically I am calling uhubctl repetitively in a loop (with no delay), and all it good until i plug another USB device, then uhubctl does the seg fault.

mvp, 2024-12-03:
I have not noticed that. uhubctl is using libusb library get information about all USB devices and control their power when possible. If you observed a crash in libusb.so, it most likely means you hit a bug in libusb. To fix this it would be very helpful to get a core dump. Core dump is essentially a special file typically having extension .core. most Linux distributions allow enabling saving core dumps in /var/core directory, read more here https://stackoverflow.com/questions/17965/how-to-generate-a-core-dump-in-linux-on-a-segmentation-fault.
If you can get core dump, I recommend opening it with gdb -c and typing `bt` (back trace) - this will provide crucial information about what caused the crash.

loupiote, 2024-12-04:
thanks. i'll let you know next time i see a core dump.

mvp, 2025-01-18:
Closing due to inactivity. Please reopen or comment if you can provide crashdump and/or stack trace. Thank you @loupiote!

closed 2025-01-18 by mvp