mvp/uhubctl #603: uhubctl much slower to cycle a hun when hub connected using USB 3.0 compared to USB 2.0, why?

loupiote, 2024-12-03:
I noticed that uhubctl is much slower at cycling my StarTech hub when I connect it on a USB port that supports USB 3.0 handshake

When I connect the same hub to the same USB port, but using a USB extension cord that only has the internal wiring to support the USB 2.0 handshake, the hub cycling takes only 0.61 sec, compared to 1.47 sec when using USB 3.0

Any idea why? Is USB 3.0 driver stack very inefficient??

Timing using USB 2.0:

$ time ./host-scripts/reset-hub.sh
+ echo hub restarting with uhubctl...
hub restarting with uhubctl...
+ uhubctl --action cycle --delay 0
Current status for hub 1-1 [14b0:013f StarTech.com USB 2.0 HUB, USB 2.10, 5 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0103 power enable connect [2581:3b7c Ledger Ledger Wallet 0000]
  Port 4: 0100 power
  Port 5: 0000 off
Sent power off request
New status for hub 1-1 [14b0:013f StarTech.com USB 2.0 HUB, USB 2.10, 5 ports, ppps]
  Port 1: 0000 off
  Port 2: 0000 off
  Port 3: 0000 off
  Port 4: 0000 off
  Port 5: 0000 off
Current status for hub 1-1 [14b0:013f StarTech.com USB 2.0 HUB, USB 2.10, 5 ports, ppps]
  Port 1: 0000 off
  Port 2: 0000 off
  Port 3: 0000 off
  Port 4: 0000 off
  Port 5: 0000 off
Sent power on request
New status for hub 1-1 [14b0:013f StarTech.com USB 2.0 HUB, USB 2.10, 5 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
  Port 5: 0100 power
+ echo hub restarted with uhubctl
hub restarted with uhubctl

real	0m0.618s
user	0m0.004s
sys	0m0.019s

Timing using USB 3.0:

$ time ./host-scripts/reset-hub.sh
+ echo hub restarting with uhubctl...
hub restarting with uhubctl...
+ uhubctl --action cycle --delay 0
Current status for hub 2-1 [14b0:013e StarTech.com USB 3.1 Gen 2 HUB, USB 3.20, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Sent power off request
New status for hub 2-1 [14b0:013e StarTech.com USB 3.1 Gen 2 HUB, USB 3.20, 4 ports, ppps]
  Port 1: 00a0 off
  Port 2: 00a0 off
  Port 3: 00a0 off
  Port 4: 00a0 off
Current status for hub 1-1 [14b0:013f StarTech.com USB 2.0 HUB, USB 2.10, 5 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0103 power enable connect [2581:3b7c Ledger Ledger Wallet 0000]
  Port 4: 0100 power
  Port 5: 0100 power
Sent power off request
New status for hub 1-1 [14b0:013f StarTech.com USB 2.0 HUB, USB 2.10, 5 ports, ppps]
  Port 1: 0000 off
  Port 2: 0000 off
  Port 3: 0000 off
  Port 4: 0000 off
  Port 5: 0000 off
Current status for hub 2-1 [14b0:013e StarTech.com USB 3.1 Gen 2 HUB, USB 3.20, 4 ports, ppps]
  Port 1: 00a0 off
  Port 2: 00a0 off
  Port 3: 00a0 off
  Port 4: 00a0 off
Sent power on request
New status for hub 2-1 [14b0:013e StarTech.com USB 3.1 Gen 2 HUB, USB 3.20, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 1-1 [14b0:013f StarTech.com USB 2.0 HUB, USB 2.10, 5 ports, ppps]
  Port 1: 0000 off
  Port 2: 0000 off
  Port 3: 0000 off
  Port 4: 0000 off
  Port 5: 0000 off
Sent power on request
New status for hub 1-1 [14b0:013f StarTech.com USB 2.0 HUB, USB 2.10, 5 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0101 power connect [2581:3b7c]
  Port 4: 0100 power
  Port 5: 0100 power
+ echo hub restarted with uhubctl
hub restarted with uhubctl

real	0m1.479s
user	0m0.001s
sys	0m0.032s

mvp, 2024-12-03:
USB3 hubs are always 2 hubs internally, sharing the same physical connection ports. If you want to turn power off on USB 3 port, you have to turn it off for both USB 3 and USB 2 chips. That alone should make switch performance for USB 2 only connection about 2x faster. However, when you use USB 2 only cable, USB 3 chip doesn't come into play, hub works in compatibility mode and thus switches power faster.

Now, I noticed that you are turning off and on all 5 ports of your hub, and it seems that you only want to reset ledger device? If your hub truly supports per port power switching, resetting only one port should make it 5x faster. If this hub doesn't support resetting one port, I recommend getting another hub from supported list.

Another way to make faster still is to use option -N, which skips querying USB descriptors.

If your host has Linux kernel 5.x, upgrading to kernel 6.x should be also slightly faster.

loupiote, 2024-12-04:
Thanks for the explanation.

I tried adding --location 2-1 when on USB 3.0, but it tells me that location does not exist (even though it lists it when i don't pass --location). If I pass --location 1-1, I think it works just like when I connect with a USB 2.0 cable, and definitely not any faster.

My hub does not address the ports separately, so even if i specify --port 3, all the ports gets cycle and there is no performance difference.

mvp, 2024-12-04:
Which uhubctl version you are using? Location doesn't exist was a bug on older versions.

Single port turning off everything is odd. If you get hub which supports per port power switching correctly, performance of controlling one port should be faster than multiple ports.

loupiote, 2024-12-05:
i am using version 2.6.0, that i compiled from the latest source code.

i checked with "uhubctl --version".

I don't think my startech hub supports per-port control.

mvp, 2024-12-05:
There is only one way to find out - give it a try!

mvp, 2025-01-18:
Closing as not actionable. Next time please open such questions as discussion. Thanks!

closed 2025-01-18 by mvp