mvp/uhubctl #604: RSH-A107 is NOT SUPPORTED

fredseo91, 2024-12-05:
I saw that RSH-A107 is on the list of the supported device for uhubctl. 
This is actually not true although they do tell you they have ppps when you print out uhubctl. 

If you run the OFF command on the ports, devices that actually communicates seem to go suspended, 
but if you try it with devices that use raw 5V such as ESP32 boards do not turn off at all.

Thus this hub should not be considered as true ppps supporting hub and should be taken off the list.

fredseo91, 2024-12-05:
someone already has mentioned it. 

https://github.com/mvp/uhubctl/issues/572#issuecomment-2416979033

mvp, 2025-01-18:
Thanks, removed this hub from supported list.

commit 40f10578 (2025-01-18): Remove Rosonway RSH-A107 (and ikuai A107) from supported list

There are multiple reports that RSH-A107 does not turn off VBUS:
While it is possible that some hardware revisions which are working,
looks like most don't. It is better to remove it from the list.

Closes #604.
closed 2025-01-18 by mvp